gnea/grbl #612: TMC2130 sensorless homing not finishing, constant alarm 8

tmohre, 2019-02-20:
Hi All,

I have read through multiple threads but it seems that I'm reporting something new/unseen so far.

My Setup is Eleksmaker Mana SE v3.2 + grbl1.1g, I have upgraded X - Stepper Driver, and  Y-Stepper driver (for Y1 and Y2 Axis motors) with TMC2130 + SPI and Endstop/Limit connection to Diag1 pin.

SPI and Diag Pins, as well as the stepper drivers themselves seem to work, as the axis are moving smoothly, no missed steps during jog operations.

I changed the homing process in config.h to reflect:
```
#define HOMING_CYCLE_0 (1<<X_AXIS)  
#define HOMING_CYCLE_1 (1<<Y_AXIS)
```

while X-axis homing is working as expected/desired 1st search run, then pull off, then locate run
on Y axis it just does the search run, then pull off and then issues Alarm 8

```
22:40:49.321 -> Grbl 1.1g ['$' for help]
22:40:49.321 -> [MSG:'$H'|'$X' to unlock]
22:40:53.996 -> [echo: $H]
22:41:00.730 -> ALARM:8
22:41:01.231 -> ok
22:41:01.265 -> 
22:41:01.265 -> Grbl 1.1g ['$' for help]
22:41:01.265 -> [MSG:'$H'|'$X' to unlock]
22:41:34.227 -> <Alarm|WPos:-10.000,-10.000,0.000|Bf:15,127|FS:0,0|WCO:0.000,0.000,0.000>
22:41:34.227 -> [echo: ]
22:41:34.227 -> ok

```
The issue stays with the Y-Axis even when switching stepper motors between X and Y axis, so both Steppers and Limit for Homing works (at least for X-Axis)

Your feedback would be greatly appreciated, as -honestly- I am currently not knowing what else to do besides just skipping the sensorless homing feature of the stepper drivers.

sinfocomp, 2019-03-13:
I have a simmilar issue, 
x axis seeks home sw and then retracts acordingly but Y axis seeks for only a few mm then issues alarm,
if the machine is at home position (home sw engaged) then it homes correctly both x and y
any help would be greatly apreciated
Thanks

Booradley98, 2019-07-29:
I don't have any help with the current issue, but I could use some help myself if that's okay! I am using the same board EleksMaker Mana SE v3.2 with Arduino Nano and A4988 Drivers, running grbl v1.1g. I am just trying to get traditional (endstop) homing to work but am having difficulty. I have endstops connected to D9 and D10, with a shared ground connected to ground. The X and Y motors move as desired but when I send $H to home nothing happens and I get Alarm 9 : Homing failed. I have tried modifying the config.h to only do X and Y homing and not Z, however, there was no difference in what happened. 
Thanks in advance!

MaicolDeSouza, 2020-03-22:
Dear friends. 
Do you know if TMC2130 with SPI control is already implemented on GRBL? 
Thanks in adavance.