gnea/grbl #613: laser v 1.1

Alive44, 2019-02-22:
Hello every one the laser dont turn off when i use the grbl v1,1  
but was work normally with v 0.9

JBSchueler, 2019-02-22:
A bit more information than "it doesn't work" would be nice.

Could you please provide more information about your cnc board?
Did you check the pin out, is it different as the default pin out of grbl?
Is your PWM output correct? Maybe your board expect an inverted signal.

If you don't have  clue about your board, please provide a picture.

Alive44, 2019-02-22:
thanks for your time . my cnc board it v3.1 Arduino  shield . Im use the (Spn En) pin to out the signal for my laser . 
note: my laser is PWM .

JBSchueler, 2019-02-25:
The `Spindle Enable` pin can not be used as PWM out pin.
You have to use pin 11 for this and Enable PWM.

When using PWM `Spindle Enable` changes to "Variable Spindle PWM" and that's why `Variable Spindle PWM` and `Limit Z-axis` have been swapped

**PWM DISABLED**
[image]

**PWM ENABLED**
[image]

I guess you have connected your laser to pin 12 which is `Limit Z-axis` and has a pull-up enabled.

For more information have a look at the [WiKi](https://github.com/grbl/grbl/wiki/Connecting-Grbl)
Please check your [cpu_map.h](https://github.com/gnea/grbl/blob/master/grbl/cpu_map.h) file  for pinning and you [config.h](https://github.com/gnea/grbl/blob/master/grbl/config.h) file if `VARIABLE_SPINDLE` is defined.