gnea/grbl #615: Laser breifly turns on when power is applied and also when usb is connected

atsage, 2019-02-26:
I read a few issues spread out around here but nothing software is helping me.  Before I go and solder a resistor on the laser pin I just wanted to create an issue to see if I am missing something.  

When I apply power to or connect the usb cable to my woodpecker board the laser turns on for some time and burns whatever I have under it.  I finally managed to flash the latest image with my USBasp to get rid of the bootloader, and that definitely helps as the laser is on for a lot less time now, but it still turns on and still burns anything I have under it.  The last suggestion I have left to try is soldering a 100k resistor to pin 11 to prevent it from floating during resets but I would like to keep that as a last resort.

chamnit, 2019-02-26:
There is a faq about this in the wiki. In short it’s due to the boot loader.

closed 2019-02-26 by chamnit

atsage, 2019-02-26:
Does the precompiled file [here ](https://github.com/gnea/grbl/releases) contain a bootloader?  I flashed this file using a USBasp so unless it has the bootloader included I don't have one anymore.

chamnit, 2019-02-26:
Sorry. There isn’t much I can do from a firmware standpoint. Most people come up with hardware solutions like always disconnecting the laser until a job a run, which is always a good idea.