gnea/grbl #616: Question about G02/G03 arcs as related to Grbl

109JB, 2019-02-26:
Not really an issue, but I'm working on a cutter radius compensation algorithm for my GUI and was wondering if there is a minimum arc radius and/or minimum arc length that Grbl can resolve.  In other words, if a really teeny tiny arc is defined, say much less than the $10 value, will Grbl choke on it, or will it just essentially make a linear move to the target in the G02/G03 line? I just need to know if I have to handle this at my gui level or if i can just pass it and let Grbl handle it.

chamnit, 2019-02-26:
Yes there is a lower limit but hard to say how small. It mostly has to do with floating point precision errors being limited to 7 digits. 

For the vast majority of arcs, single precision is just fine. But if you an axis offset or radius of something like 0.001” or less, you can start running to math issues and behavior can become unpredictable. 

Grbl doesn’t have the room or extra cycles to check for these things but these errors are very rare. Often CAM programs have a minimum target length (or something similar) which will just insert a line motion if the arc is sufficiently small. They do so because it’s just intrinsic to most CNC controller being limited by floating point math precision.

davidelang, 2019-02-26:
This precision limit will also cause problems with very large radius arcs.

David Lang

109JB, 2019-02-27:
The radius won't be large as it will be limited to the radius of whatever cutter is being used.

What this is about is when implementing cutter radius/diameter comp if you encounter an outside corner like going around the outside perimeter of a part, the comp routine inserts an arc at the corner like the picture below. The potential problem is shown on the right where the change in direction is small. The resulting inserted arc becomes shorter and shorter. If the change is really small it sounds like there could be issues if this is allowed to happen.  I'll take care of it by checking the distance between the start and end point of the inserted arc and if below a certain value I will just use the projected intersection of the straight paths and forego the inserted arc, or I'll just insert a line instead.

In typing this I realize there could be another issue with my implementation that I have to handle. Normally the cutter radius and hence the offset of the path is the actual radius of the cutter, but there is another way to do it and is convenient when manually converting a program without comp to one with comp. If the program is written without comp the path is already offset to the intended cutter radius. This however means that using re-sharpened cutters or a cutter of a different diameter will yield a part that isn't the right size. Comp can be added relatively easily by inserting G41 or G42 commands as appropriate and then entering a tool radius that rather than being the cutter radius is the difference between the intended cutter diameter and the diameter of the cutter actually used. For example, if programmed for a 0.5" cutter but using a 0.375 cutter the entered cutter diameter would be -0.125.  Where this might cause me a problem is if the actual diameter is very close to the programmed cutter diameter, such as when using a re-sharpened end mill. I could wind up with a small radius as well as a small arc length. I'll have to handle this potential small arc radius issue in a similar fashion to the small arc length. Oh well.

[image]

davidelang, 2019-02-27:
On Tue, 26 Feb 2019, 109JB wrote:

> The radius won't be large as it will be limited to the radius of whatever cutter is being used.

umm, no. A G02/G03 arch can be effectively flat over the cutting area (i.e. a 
straight line), in no way is it limited by the diameter of the bit.

even the minimum radius of the movement isn't limited by the bit diameter.

David Lang

chamnit, 2019-02-27:
I once talked to a developer that implemented cutter comp. He warned that there are a lot of weird edge cases you have to deal with. One of them is arcs. The other is disappearing features and how to handle them.

109JB, 2019-02-27:
> umm, no. A G02/G03 arch can be effectively flat over the cutting area (i.e. a straight line), in no way is it limited by the diameter of the bit. even the minimum radius of the movement isn't limited by the bit diameter. David Lang

ummm yes.  Look at what I am referring to in regard to my question. The radius of the added arc for cutter radius compensation CANNOT be larger than the cutter radius. I am not talking about G02/G03 in general but in my specific application as it applies to cutter radius compensation.

109JB, 2019-02-27:
>I once talked to a developer that implemented cutter comp. He warned that there are a lot of weird edge cases you have to deal with. One of them is arcs. The other is disappearing features and how to handle them.

Yep. It certainly is a ball of dung.  I'm breaking it up into 4 broad brush scenarios with smaller sub-cases in each. These 4 large categories are line to line, line to arc, arc to line, and arc to arc.

The disappearing features is a particularly annoying thing. Even just thinking about possible scenarios that result in this is are difficult. I'm making headway though.

109JB, 2019-02-28:
Got the answer so closing

closed 2019-02-28 by 109JB