gnea/grbl #617: limit sensors

Peter6891, 2019-02-26:
I noticed that the limit sensors work only when starting a zero search. But then, when the milling program is running and if the machine runs into the limit sensor, it does not work - the program does not know that the machine has hit the limit sensor. Tell me, is it conceived on purpose, should it work?

109JB, 2019-02-27:
You have to activate hard limits in the Grbl settings.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

Peter6891, 2019-02-27:
Ok! Thank's a lot

jflapao, 2019-03-03:
Hello, I'm using GRBL for a will and until now I never use limits sensors because I had problems with electrical noise on the connection, I have use a normal switch, a switch with integrated filters , even a proximity sensor and no one have results, I have try use another power source for the limits. I have perform the connection you say here https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches and noting, so I have tried  one thing leve the GRBL on with the limits activated on configurations
$20=1 for software limits on
$21=1 for hard limits on
$22=1 for active the homing cycles on
$23=3 to due the homing on the Z top, and the X and Y bottom left
I changed the $ 5 to 1 to use the circuit in NC, but without nothing connected to the CNC shield and after a will (I can't say the time because was on the night) by GRBL sender the bCNC was report a switch trigger, so I assume the noise comes from another source, and the filters on the switches will not solve.
I can use the machine without limits, but I want to build another one, more robust and professional, and I wanted to use the limits, to be able to use in full, because without the limits the machine accepts any g-code and goes to the physical limits of it. For this I intend to use closed loop servo step motors with integrated encoder, is there any way for the GRBL to receive the data from these encoders and correct the missing steps if any?

SailWithChips, 2019-03-03:
Hello jflapao, it has been transferred in advanced. The closed loop is something in the drive it self, not in the grbl firmware. Basically what I’m trying to transfer is that, if some stepper or servo lost some step then the drive should correct since the position control is in the drive part... no position control into the grbl... if the encoder then goes to ‘where grbl is’ visa some arduino or whatever input then, yes the position control loop should be in the controller but it is not the intention of the grbl as far as I understood from some tread from the grbl gurus... please correct me if I’m wrong

jflapao, 2019-03-04:
Hello SailWithChips what you say have a point, but I have some doubts, for
imagine I am the drive X, thou drive yo gnea the drive Z and GitHub the
GRBL controller, and we are doing a job in which we we have to move the
three simultaneously, and I lose one or two steps how do I inform you and
the gnea to wait for me to correct the lost steps and then continue? It can
only be through the GRBL GitHub controller, so there has to be some
feedback from the closed loop drive to the GRBL, so I read before, the GRBL
sends the information to the drives and its work is done, it does not wait
for any response, Or am I wrong?

We work on rocks, and we use a CNC that is both a bridge saw and a milling
machine, and because the machine is already old, whenever we want to do
fressing work, we have to harvest the saw that is time-consuming, and we
would be willing to "to construct" a milling machine for these jobs leaving
the bridge saw as bridge saw, but wanted to build something with
certainties of correct operation.

José Lapão

967007851

à(s) 17:47:

SailWithChips, 2019-03-04:
I would say that I would use standard function from servodrives... I mean: In the case of too much ‘following error’ then output a boolean alarm and with some alarm from drives in parallel then act to ‘feed hold’ funtion for instance.
How rapid should drive correct this error is something parameterizable in the drive itself via some Kp for position loop or Kp & Ki for speed loop... for instance