gnea/grbl #618: 4th axis or A axis

beckmx, 2019-03-01:
Hello guys, I was wondering if the spindle pin could be used as a 4th axis controller, I was using the gcodetools from Inkscape and the feature "tangent knife" creates an "A" command, is there some how I can attach a motor, stepper or something?

MeJasonT, 2019-03-05:
What happened to the A axis, the original GRBL GUI showed it as an option but I never knew if it had been implemented or not.

Carl-jackslabs, 2019-03-19:
Perhaps you could wire a servo with your rotational requirements as a starting point for using PWM  to control that axis. There are many better ways to do this but if you're not so sure and wanted to try it quickly a servo might be the fastest way to find out.

mikemaikaefer, 2019-04-08:
Have a look at https://sourceforge.net/projects/axis5/ , this is a GRBL-fork which supports up to 5 axes - but then of course requires a MCU with more resources, so it will probably not work with the 328p.