gnea/grbl #619: Is there an example how grbl works with coreXY? Complete grbl noob

vitsensei, 2019-03-01:
Hi,
As far as I understand, grbl now has pin 11 which is a PWM pin, free to use. I assume that this can be used to control a servo (pen up/down movement).
But how does it actually work? For example, what sort of command should I send?

Side note: Is there any low level gcode generation from a photo? I know InkScape has this feature but it has too many other extra features that I don't necessary use. I'm thinking something low level like a Python script.

MARIOBASZ, 2019-03-02:
I think that from any program or interface I think it says, you can edit the gcode. I use BCNC, it has interesting add-ons: I generate all the routes from there. I think there is something to generate from a photo. Use phyton. Some contributions to the soft I do. It would be good if you find it useful and also want to make contributions.

swarfer, 2019-03-02:
I have used this a bit.   It does a good job but you do have to experiment with image size and other settings to get good results. 
http://www.scorchworks.com/Dmap2gcode/dmap2gcode.htm

Pin 11 pwm is usually for controlling the speed of the spindle. Do you want servo control as for an Eggbot?

vitsensei, 2019-03-03:
One more question. How can I change the step/mm setting when use corexy?
I have enabled corexy in config.h and then changed DEFAULT_X_STEPS_PER_MM value (did the same with Y and Z too) in DEFAULTS_GENERIC, but it does not seem like it does anything.
In my config.h, it has #define DEFAULTS_GENERIC line, so I guess I'm changing the right setting?

Thanks for the help.