gnea/grbl #620: Woodpecker - Laser 3 pin wiring

dolfek, 2019-03-01:


dolfek, 2019-03-01:
Hi
I'm  upgrade 3018 Woodpecker to grbl 1.1. But have  a problem with laser. How can connect 2.5W 3 pin module to Woodpecker 3.1.?
It have just 2pin spindle, 2pin laser outputs and some 12V 2pin.
I'm connecting + and - from laser to laser separate power suply but pwm I dont know where to connect.
With lasergrbl it's come just the andjustment light on. When try demo NC code all axis work fine but laser  not burn.
Help please and sorry abaut bad english.

dolfek, 2019-03-03:
Is anybody out ther?

JBSchueler, 2019-03-04:
This is more a hardware related isue and has nothing to do with GRBL.
If you connect your laser toa separate power supply, you only have to connect the PWM TTL  and ground from the board to your laser. Connect ground from your board to the ground of the power supply of your laser.
Otherwise it will not work at all...

More I can not help you as I have not that board.

dolfek, 2019-03-04:
Thank you for reaplay. I have  separate power supply and 2 pin laser , 2 pin spindle output.
I connected pwm and gnd from laser to laser pins but its just adjustment light is on. If press a switch on laser burning at full. I can't adjusting power. If pwm connecting to z end pin is the same. gbrl 1.1 have swaped pins z and spindle. But where can find pin 12 on board?

dolfek, 2019-03-04:
Hi 
I solved. Solder pwm connection to bottom pin on MOSFET over 10k resistor, Vcc and Gnd connect on laser output (2pin) without external PWS. It's work. 
Now is time to go forward. Must build an enklosure, protect wires from noises, fine adjustment of laser, ... .
Now, when all work, I can play with finese.

My configuration: 3018 Cnc, Woodpecker 3.1, laser 2.5W cheap China clone, all endstops, grbl 1.1, Candle 1.17, Lasergrbl, spindle 775 motor with collets, ... .
I'm happy with begining reasult, after four months and work further.
On end is all simple.

alienboyxp, 2019-10-11:
Hello @dolfek  I have similar setup and I have troubles to set up my laser correctly. Can you please share some images about your setup/connections can help me to identify my issues?

dolfek, 2019-10-11:
My pictures:
 IMG_20191011_141825.jpg
<https://drive.google.com/file/d/15fjxqsCxkaTbzSrCWoauLEmIoZtcMF68/view?usp=drive_web>
 IMG_20191011_141833.jpg
<https://drive.google.com/file/d/1JntVDuzj2gHABiWsKa3i6vVz-G7Gi95J/view?usp=drive_web>
 IMG_20191011_141957.jpg
<https://drive.google.com/file/d/1L5PaJfBXof1DyDmq3X6ekWyhOzeGkVmD/view?usp=drive_web>
 IMG_20191011_142029.jpg
<https://drive.google.com/file/d/1qawmGhPhuzNDOAJO97-8-GDEkrMmH9XF/view?usp=drive_web>
 IMG_20191011_142054.jpg
<https://drive.google.com/file/d/1GytCYx7oVpBBBoncNHcX30kqcmryYttX/view?usp=drive_web>

alienboyxp, 2019-10-11:
Thanks @dolfek my setup is a bit different, I have 3 pins on my Woodpecker board and my laser have 2 different 2 pin connectors... one for 12v and ground (should be connected to external power) and the other for TTL/PWM (tagged as TTL only but not sure if can manage PWL). I test different connections but no luck to make it work as expected.
Any help will be welcomed :D

dolfek, 2019-10-11:
You have just reversed as I'm. Look at github.com/gnea/grbl page issue.
I'm not sure, but my original 3 wire settup must be: black and red on power
and white(yellow) to woodpecker laser output. GND must be connected
together.

Edoggy2, 2023-11-15:
> Thanks @dolfek my setup is a bit different, I have 3 pins on my Woodpecker board and my laser have 2 different 2 pin connectors... one for 12v and ground (should be connected to external power) and [...]

Goodmorning, I'm having the same situation! Did you find any solutions? If so, can you share the details? thanks

lalo-uy, 2023-11-16:
Did you try to put a ttl signal between TTL input and ground? Say a simple
50/50 clock?