gnea/grbl #622: Parts came out with wrong dimension

andreasaditya90, 2019-03-04:
so i tried making this part with my cnc router. I designed it with fusion 360. it was made with 2 separate contouring operation. one for the inside circle, and the other one for outside profile
[image]

however, the part came out look like this
[image]
The circle was not in the middle

i thought there's mechanical issue with my machine. i checked everything, and its fine.
so i tried instead of separating the two operation, i combine them in one operation.
the part look like this
[image]

the result is somewhat acceptable. i dont know what im missing in here.
hopefully someone will help me solving this problem.
thank you very much

langwadt, 2019-03-04:
you picked a different origin in fusion for the second op?

andreasaditya90, 2019-03-04:
> 
> 
> you picked a different origin in fusion for the second op?

thank you for your prompt reply. im sure i didnt picked different origin. The main problem is if i separate the two contouring operation ( inside circle and outside profile), the parts came out wrong. But if i combine them into one contouring operation, the parts came out fine.

n0th1n6, 2019-03-04:
In the manufacture, how many setup do you have?

andreasaditya90, 2019-03-04:
> 
> In the manufacture, how many setup do you have?
> […](#)
> -- *--------------------------------* *Noli Rafallo*

only one setup

andreasaditya90, 2019-03-04:
i just found out my workpiece move slightly during milling process and cause this problem.
so the problems has been found!

thank you very much for all the replies

Ben3styx, 2019-04-06:
Sometimes there just aint enough clamps.