gnea/grbl #628: camtool cnc v3.3 rx tx

fdman, 2019-03-21:
Hello,

I have this new board and i would like to connect this card wifi with esp32 but I do not know where to find the rx tx.
Is it possible to connect the rx tx on the ofline grbl controller ?
Where can I find the rx tx ?

Thanks

[image]

ASHIT-AXAR, 2019-03-23:
it looks there are no pinouts for Rx and Tx.
in the bottom left of the picture there is a IC with 16 pins (called: CH340G) 
i think you need to solder 3 wires to the IC to take out the serial communication functionality yourself.
you should count pin numbers from top right side to the left.
pin1: GND
pin2: Tx
pin3: Rx
then you can connect these 3 wires directly to your bluetooth or wifi adapter.

fdman, 2019-03-24:
that's what I thought I'd do, I'm going to test

thanks

MeJasonT, 2019-03-24:
the CH340G is a serial to usb ic
[https://cdn.sparkfun.com/datasheets/Dev/Arduino/Other/CH340DS1.PDF](url)
there is a chance that what you are trying to do wont work unless you remove the CH340G ic.  it will work by pulling up the transmission lines and will interfere with the signals you are trying to piggyback onto them.  You might end up having to butcher the board to get wifi to connect and work. I don't know if just disconnecting the power pin to the CH340  will do the trick or if its full removal would be needed.
check there isn't a resistor in the rx /tx path instead that you could remove and solder your wires on there - you can then pick up a ground from anywhere, it should be common across the board.
a schematic will be invaluable if you have one.
Id be interested to see how it works out.

Fusseldieb, 2019-04-09:
Take note: This version of GRBL doesn't support the ESP32. [This one does.](https://github.com/bdring/Grbl_Esp32).

Furthermore, passing G-Code over Wi-Fi isn't necessarily a good idea. If the Wi-Fi isn't stable, it might delay some commands, stop mid process or worse. Don't take Wi-Fi reliability for granted. 

Since your board seems to have "LASER" written on it, it might be a Laser CNC board, which means that if the connection isn't stable enough, it would just stop mid-process and burn a hole into god-knows-what material you're doing at that time. Not good, in my opinion.

daveb693, 2019-04-17:
i have same board does anyone know how to set up endstops on it. info is very hard to find on this board
thanks

hydra, 2019-04-20:
I've got one too, it comes with the CNC 3018 Pro.  Any schematics or related info would be very useful.

There were schematics for the Woodpecker board here, as anyone investigated if this is just the same board but laid out differently?

http://linksprite.com/wiki/index.php5?title=DIY_CNC_3_Axis_2418_Engraver_Machine_PCB_Milling_Wood_Carving_Router_Kit_Arduino_Grbl

KimPedersen, 2019-04-29:
They are available on the 8 pin connector at the bottom left. Top row from left to right is 5v, ground, TX, RX. Bottom row from left to right, 5v ground, reset, and the last pin I haven't figured out, maybe a data pins that goes high when the mega328p is done booting, to indicate whatever is plugged in can boot now.

1afc, 2019-06-14:
@daveb693 
I'm a noob at this but learnt that the last three pairs of pins on the bottom left header strip are for Z Y and X end stops. 
The board has to be told to make them active with $21 = 1 (on) and 0 (off).
Once set that will be the default on power up.

1afc, 2019-06-14:
CAN ANYONE HELP WITH LASER ACTIVATION
I am not using my V3.3 cnc for laser or routing so there should be no clashes with mechanical things but I am interested in activating other things with the Laser A and Laser B plugs.
AFAIK they run as a spindle and the code to turn them on is written on the back of the board.
S-=0   Laser 1 on
S-=1   Laser 1 off
S+=0  Laser 2 on
S+=1  Laser 2 off

I also understand they have to have a Speed >0 and be in G1 or other movement codes.

Can the spindles and lasers be on at the same time and if not, what codes are needed to turn the spindle output off and turn on the lasers.

miguelteam, 2019-06-18:
Hello
Can someone help me, I have a problem with the connection of a laser with camtool cnc v3.3 motherboard, I have installed a 15 w laser and it does not turn on, I have connected it to laser "b" but I do not know, what parameters to give it in the code?, I have tested and the laser work directly with 12 v

sorry for my english, I'm using the google translator
Thank you