gnea/grbl #629: WoodPecker and CronosMaker cards not communicate with usb and offline controller

erdemarslan, 2019-03-21:
Hi everyone,
I am newbee on cnc and grbl. I buy CNC3018 Pro. It come with Woodpecker 3.2A control card. I connect probe pins to A5 and GND. When i touch probe GND to spindle, spindle runs. Now my card not communicate with usb and offline controller. I think one or more components burned but i dont solve that.

I buy CronosMaker CNC Controller card from AliExpress. This is link: https://www.aliexpress.com/item/CNC-3018-2418-1610-3-Axis-control-board-GRBL-Offline-Controller-board-Laser-board-for-GRBL/32866461881.html?spm=a2g0s.9042311.0.0.35514c4dITool4

Today, i connect probe pins and when i touch probe's GND cable to spindle, spindle runs. Now this card not respone to usb and offline controller.

What i make wrong? Can i repair my cards? Anybody help me? Thanks. Sorry for my English.

kris0377, 2019-04-25:
You need to replace CH340 chip

erdemarslan, 2019-04-25:
I try it but nothing changed.

kris0377, 2019-04-25:
Atmega chip replasement doesn`t help too?

erdemarslan, 2019-05-18:
> 
> 
> Atmega chip replasement doesn`t help too?

I changed atmega chip but not solve my problem.

kris0377, 2019-05-18:
You should flash it by using programm device or arduino uno board.

quotelawrence, 2021-01-20:
I just purchased the exact same board I have traced an issue that the ground is Hot and the Positive is grounds, i tried wiring it so that hot is going into the laser module correct, and the ground is correct, and that Blue PWM is matched, i merely plug in the usb no power and the laser is on and the fan is not and i have kept it while hitting the on off function there is no communicating, this board say grbl 1.1f but I have other that say the same 1.1f and I am having trouble with them all