gnea/grbl #631: GRBL1.1 Not compiling - Exit Status 1 Error compiling for board Arduino Nano.

kleintoren, 2019-03-23:
Hi. I get "Error compiling for board Arduino Nano." error on fresh install of Arduino IDE
Grbl1.1f latest download

Full Error message below:

Arduino: 1.8.9 (Windows 10), Board: "Arduino Nano, ATmega328P"

In file included from C:\Users\oddw2\Documents\Arduino\libraries\grbl\grbl.h:46:0,

                 from C:\Users\oddw2\Documents\Arduino\libraries\grbl\settings.c:22:

C:\Users\oddw2\Documents\Arduino\libraries\grbl\settings.c: In function 'settings_restore':

C:\Users\oddw2\Documents\Arduino\libraries\grbl\defaults.h:519:37: error: expected expression before '=' token

   #define DEFAULT_SOFT_LIMIT_ENABLE = // false

                                     ^

C:\Users\oddw2\Documents\Arduino\libraries\grbl\settings.c:93:9: note: in expansion of macro 'DEFAULT_SOFT_LIMIT_ENABLE'

     if (DEFAULT_SOFT_LIMIT_ENABLE) { settings.flags |= BITFLAG_SOFT_LIMIT_ENABLE; }

         ^

Using library grbl in folder: C:\Users\oddw2\Documents\Arduino\libraries\grbl (legacy)
exit status 1
Error compiling for board Arduino Nano.

kleintoren, 2019-03-23:
There is an "=" instead of a "0" in defaults.h on the latest download on line 519

closed 2019-03-23 by kleintoren

MeJasonT, 2019-03-23:
I spotted a new issue uploading to a nano.  The new version of Arduino IDE has two versions of compiler for the ATMEGA328 IC .  It had me stumped for a wee while.

[image]