gnea/grbl #632: Mega 2560 will not connect

mrtwizta, 2019-03-26:
I get this error when trying to open with Gbrl Controller
No data from COM port after connect. Expecting Gbrl version string

mrtwizta, 2019-03-26:
Is there a Version of gbrl that has been tested on 2560 with out issues?

akrpkv, 2019-03-28:
https://github.com/gnea/grbl-Mega

mrtwizta, 2019-03-28:
tried the At mega 2560 and and MKS Gen2-Z (2560 chip)
"No data from COM port after connection.Expecting Grbl version string."
So I loaded A Marlin printer sketch i use for my 3Dprinter and it works fine, obviously till the heater hardware reports an error.
Just to confirm/prove the hardware and wiring is correct, i can move and home all axis without problems
so it seems the devs need to do some soul searching there?

mrtwizta, 2019-03-28:
> https://github.com/gnea/grbl-Mega

There is no response from any of the 2560 builds? they all unresponsive or error prone
most report gbrl no finished loading or no data from com port
why dont they just take a Marlin by the tail and mod it work with gbrl?
has all the bells, even the display part for jogging works?

matonym, 2019-03-28:
I'm running GRBL on the Mega 2560 without any problems.
Have you tried connecting with a terminal program to the COM-port (115200 baud) ?

mrtwizta, 2019-03-28:
how do i do that? noob here

matonym, 2019-03-29:
There's plenty of software. If you are on Windows you might try Putty (https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html)

Example
[image]

Settings. Make sure you have the right COM-port etc.

[image]

Result (Hopefully)

mrtwizta, 2019-03-29:
hi
thanks for the mail, it is appreciated.
i followed that eg. and the terminal just shows a green cursor with no text

Regards

Michae lk
michael k (under cover matter-control pimp)
+27 63 185 0233

3rdzeropoint@gmail.com <michaelkruis58@gmail.com>

matonym, 2019-03-29:
Are you sure it's the right COM-port selected?
If not sure, double-check it in the device manager.

MeJasonT, 2019-03-29:
From your earlier post it sounds like you mat not have actually installed GRBL.
in the Arduino IDE under Tools tab see if you mega is connected, under the same (Tools) tab check that the board you have selected is The Genuine mega.  Go back into Tools menu and select Serial Monitor - when it opens check the baud rate is set for 115200 in the bottom left corner of monitor screen, if not use the drop down to select it.  GRBL should have responded if not in the command bar at the top enter $$ and press the send key to the right.  If you get no response then the program on the board is not GRBL.  in that case close monitor window and go to File - Examples and open Basics - Analogue ReadSerial, load it to your board.  Open the monitor window again change the baud rate to 9600 and you should see a list of values being printed.  Your board is working and communicating if this works.
If you have installed the GRBL library on your machine go to File tab- open examples again and look for GRBL once open upload it to your mega.  open the serial monitor again, set baud rate to 115200 and see if the board responds if not try sending $$ again.  It is possible to blow up the micro or the FTDI IC but the steps above should give you some confidence that your board is working.

mrtwizta, 2019-03-29:
yes its the right one, as i said i loaded a marlin sketch that works on the
same port, used matter control to move and check works fine with matter
control and marlin
i can connect to the port with the software but gives those errors no data
or not gbrl finished loading
Regards

Michae lk
michael k (under cover matter-control pimp)
+27 63 185 0233

3rdzeropoint@gmail.com <michaelkruis58@gmail.com>

MeJasonT, 2019-03-29:
Have you installed the GRBL library correctly ?

MeJasonT, 2019-03-29:
If you click on grbl outside of the library folder it opens and looks like it should, however when you go to upload firmware it will tell you lies as it cant find any of the associated files and will fail to upload.

mrtwizta, 2019-03-29:
Regards

Michae lk
michael k (under cover matter-control pimp)
+27 63 185 0233

3rdzeropoint@gmail.com <michaelkruis58@gmail.com>

MeJasonT, 2019-03-29:
?
have you fixed it, if so please close this issue.

mrtwizta, 2019-03-29:
Be patient please i am still trying to go through your mail
Regards

Michae lk
michael k (under cover matter-control pimp)
+27 63 185 0233

3rdzeropoint@gmail.com <michaelkruis58@gmail.com>

MeJasonT, 2019-03-29:
Sorry buddy you responded with just Regards, it sounded positive.

mrtwizta, 2019-03-29:
NP
ok i did the AnalogRead and i see lots of 0, apearing

MeJasonT, 2019-03-29:
your board is communicating so no hardware fault, It looks like you are not getting the GRBL firmware loaded properly.  What machine are you wanting to use with grbl 3 axis ?

mrtwizta, 2019-03-29:
ok i did this and found the gbrlUpload under examples which is the one i have been using,
but i get nul output on serial mon
If you have installed the GRBL library on your machine go to File tab- open examples again and look for GRBL once open upload it to your mega. open the serial monitor again, set baud rate to 115200 and see if the board responds if not try sending $$ again. It is possible to blow up the micro or the FTDI IC but the steps above should give you some confidence that your board is working.

MeJasonT, 2019-03-29:
Do you want me to talk you through the install

mrtwizta, 2019-03-29:
yes please

mrtwizta, 2019-03-29:
i wish they would allow team viewer in these situations

mrtwizta, 2019-03-29:
yes please
i am in git as well i can reply there if you like?

Regards

Michae lk
michael k (under cover matter-control pimp)
+27 63 185 0233

3rdzeropoint@gmail.com <michaelkruis58@gmail.com>

MeJasonT, 2019-03-29:
Ok, 
have you located the arduino library folder
do you have any versions of grbl in your arduino library folder if so delete them, lets start from scratch.
You have two options of uploading the file by using the arduino ide to upload the program or by uploading the hex file directly using Xloader - which are you happiest using.

mrtwizta, 2019-03-29:
ok

mrtwizta, 2019-03-29:
i have a hex and i di do that but let me delete from the lib

MeJasonT, 2019-03-29:
i was going to suggest a link you beat me to it. send the link you used to download the master

mrtwizta, 2019-03-29:
https://github.com/gnea

MeJasonT, 2019-03-29:
OK tops

MeJasonT, 2019-03-29:
So you have chosen to download the GRBL program via the download zip, correct.

mrtwizta, 2019-03-29:
C:\Program Files (x86)\Arduino\libraries
is this the correct place for the library?

mrtwizta, 2019-03-29:
yes

MeJasonT, 2019-03-29:
Ummm what version of arduino IDE are you using. the latest versions save the library to another location - by default its in your document folder

mrtwizta, 2019-03-29:
1.8.9

mrtwizta, 2019-03-29:
you want to connect to my pc with team viewer?

MeJasonT, 2019-03-29:
1% newer than mine, i'm still on 1.8.8
would prefer not to use team viewer if its all the same, we have dodgy companies here in uk trying to gain access to users machines, best to stay away from it.

mrtwizta, 2019-03-29:
ok

MeJasonT, 2019-03-29:
do you have a folder like C:\Users\Your Computer\Documents\Arduino\libraries

mrtwizta, 2019-03-29:
yes

MeJasonT, 2019-03-29:
Ok is the grbl folder the top level or does it contain a folder called grbl inside it

mrtwizta, 2019-03-29:
its top, i see all the c and h files init

MeJasonT, 2019-03-29:
you are a star, now open the ide and open grbl from the examples tab

MeJasonT, 2019-03-29:
It will open just one page, but it now knows where to find the other code segments

mrtwizta, 2019-03-29:
i only see grblUload there, but its done

MeJasonT, 2019-03-29:
you should now be ready to go.  open the serial monitor window in the ide baud 115200 and send $$

mrtwizta, 2019-03-29:
done but still no response

MeJasonT, 2019-03-29:
NO!, did you get any errors when you uploaded to the board

mrtwizta, 2019-03-29:
no

MeJasonT, 2019-03-29:
im going to the workshop for an mega board, give me 10 mins

mrtwizta, 2019-03-29:
i need a brew....

mrtwizta, 2019-03-29:
ok
mine is not a Genuine arduino a clone but atm i have the other MKS Gen2 wich is a mega based board? must i try again with the Mega clone?

MeJasonT, 2019-03-29:
I have a clone as well

mrtwizta, 2019-03-29:
i have no issues with them, i run my printers on them

MeJasonT, 2019-03-29:
Do you have folders inside your grbl folder saying  build, docs, grbl 
The marlin has been written to display the full folder hierarchy grbl has a .h file which fetches all the sub files into the application

mrtwizta, 2019-03-29:
no, but i know what you talking about, i have seen that in other versions

MeJasonT, 2019-03-29:
I have just loaded my ide/grbl and it was missing after just telling you.

mrtwizta, 2019-03-29:
there are 4 files there 1 COPYING, 2 Makefile, 3 .gitignore 4 README

mrtwizta, 2019-03-29:
so the file in git is incomplete?

MeJasonT, 2019-03-29:
ok.  The zip will have contained the full project in the folder there is a folder called grbl inside it has all the .c and .h files. this is the only folder which should be in side your library not the rest of the guff

[27 more replies]

closed 2019-03-29 by mrtwizta