gnea/grbl #633: use Z axis for another Y axis

wuyoutech, 2019-03-30:
I'm building a two-Y-axis system. I wonder if there is a simple way to use the Z axis of GRBL for another Y axis. They should have exactly the same behavior, except Homing.

I know that dividing the wires into two is a solution. But that means they can't homing separately. I hope there is a better way to solve this problem.

flug, 2019-04-03:
if you use an arduino uno you can clone with A driver moteur

Fusseldieb, 2019-04-09:
Well, just send the same command to Y and to Z and it will move exactly the same way.

senod, 2020-11-24:
Ciao, anch'io ho pensato la stessa cosa ho una scheda CAMXTOOL V3.4. Visto che hai fatto la domanda tanto tempo fa spero che sei riuscito farlo in qualche modo @wuyoutech. Ma io non mi intendo tanto con arduino. Poi come dice @Fusseldieb potrei farlo così ma per fare questo dovrei mettermi copiare tutte le referenze ottenute con gcode per la asse y e scrivere gli stessi valori per la asse z. Ma esiste qualche modo per farlo in automatico? Grazie
[image]

beamzer, 2022-11-28:
I came here for the same question. Would be nice if someone who understands the grbl code framework would work out how to enable this. Normally it wouldn't be to hard to do it in hardware, but i have the same board as above and it's all SMD so not so easy to disconnect STEP and DIR for Z and solder them the X or Y output.