gnea/grbl #636: GRBL on ARM

mikemaikaefer, 2019-04-08:
When I search for GRBL for ARM-variants of the AVR, I end up either with closed requests or discussions that never came to an end.

So my questions: is there a port available for an AVR-ARM CPU? If yes, where can the sources be found and what hardware/board is it running at exactly?

GRBL is nice but the current maximum step frequency is far away from being competitive comparing to other projects and open hardware...

shooter64738, 2019-04-10:
Theres not an 'official' port of it that Im aware of, but there are grbl arm ports running on stm,esp,  and due boards out there. A few running on ti boards. Most of the arm ports I am aware of run in the 120-150khz step range. 

Also I wouldn't look for AVR-ARM cpu's it may limit your results. The AVR is a different family than the ARM architecture cpu's.

chamnit, 2019-04-10:
Grbl was never intended to support everything, but to make CNC widely available with a low barrier for entry. It’s 30kHz step rate limit is fine for majority of applications. That said, the next version of Grbl is being worked on. Don’t worry about that. It’ll be released when ready and it supports much faster arm processors. The samd21 as it’s baseline. Other processors are easy to add. 

Fortunately, I do have some good news. I received authorization to continue working on Grbl just this last week from my new job, which hasn’t been the case since mid January.

closed 2019-04-10 by chamnit

brave-ulysses, 2019-04-22:
mike - i used this to control my minilathe
https://github.com/Schildkroet/GRBL-Advanced

good luck, clay