gnea/grbl #637: how to set spindle speed?

J-Dunn, 2019-04-16:
Hi,  I am trying to set up a PWM output for spindle speed control.  I find `#define VARIABLE_SPINDLE` but can't find how to set the speed. 

G97 would seem to be the GCODE but GRBL does not list this. 

I've been ploughing through the source but have not managed to work back to a command which does this.  

Can someone help?   TIA.

mschorer, 2019-04-16:
Hi,

It's "Sxxx" like speed ;-) xxx Adair ranging from 0/off to 1000/full speed.
Combined with M3/M4 obviously ...

markus schorer mobile

J-Dunn, 2019-04-16:
Many thanks for a speedy reply.  I had found the parsing of S but could not work out what it was related to. I thought M3, M4, M5 where just setting spindle STATE, not speed.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

> Spindle State | M3, M4, M5

That is the only occurrence of M3 on that page.

109JB, 2019-04-16:
M3 S1000 would set CW rotation at 1000 rpm

Not sure how it is in the code but PWM duty cycle is values 0-255, so if max RPM setting is 1000 rpm then setting M3 S1000 should set the duty cycle to 255. Likewise M3 S500 should set the PWM duty cycle to 127. The max RPM is set in EEPROM settings

J-Dunn, 2019-04-16:
You are correct the scale (on Uno) is 0-255; however this does not have to be 1000 rpm.  It is defined by DEFAULT_SPINDLE_RPM_MAX

shapeoko sets it to 10,000 ; sherline 2800 and of course you set it to whatever your spindle can do. 

However, M3 S1000 is 1000 rpm , not "full speed". I just wanted to correct that in case it misled anyone reading this in the future.  Thanks to all.

closed 2019-04-16 by J-Dunn

109JB, 2019-04-17:
> You are correct the scale (on Uno) is 0-255; however this does not have to be 1000 rpm. It is defined by DEFAULT_SPINDLE_RPM_MAX

Please read the post before claiming it is in error. I never said that the only ma. Rpm setting was 1000 rpm. What I said was:

"so if max RPM setting is 1000 rpm..."

And, "The max RPM is set in EEPROM settings."

Both of these are correct and you don't have to get into the actual c coding to get to either.

J-Dunn, 2019-04-17:
Chill out JB, the main thing I wanted to correct was the initial statement from markus, as I stated in the last sentence of the post which you are complaining about. 

>  It's "Sxxx" like speed ;-) xxx Adair ranging from 0/off to 1000/full speed.

> However, M3 S1000 is 1000 rpm , not "full speed". I just wanted to correct that in case it misled anyone reading this in the future. Thanks to all.

min.rpm and max.rmp are indeed in the EEPROM settings and available through $30 and $31.
The initial value is hardcoded in DEFAULT_SPINDLE_RPM_MAX

Thanks for your comments.

mschorer, 2019-04-17:
Hi,

To clarify what I meant:
Giving S<max> will result in an always/fully "active" pwm pin - which will
result in full speed.
Max may be 255, 1000 or even adjustable - but there is no way of knowing
how fast the spindle is actually running.
It might be a good idea to adjust max to your spindles top speed though -
to not require scaling.

markus schorer mobile

J-Dunn, 2019-04-17:
> fully "active" pwm pin - which will result in full speed.
It may not be full speed if your controller responds to a 0-10V input. As mine does. 

> but there is no way of knowing how fast the spindle is actually running.

The idea is to match the voltage produced by 255 to fit the control voltage of the spindle controller by using $30.   If 100% pwm gives 5V and your controller regulates at 800 rpm when it gets an input of 5V, then you inform GRBL of this by setting $30=800 ( or building with DEFAULT_SPINDLE_RPM_MAX 800 ). 

If fact my controller supplies 10V output which I feed back via an opto-coupler to protect the GPIO running PWM, so it gets input range like 0.5-10V . The actual rpm produced by 10V i/p is then configurable via the controller's parameters. Once I have chosen the top RPM I want for the machine ( ie what it will do give a 10V signal ), this gets set via $30. It is a PID controlled variable frequency driver. So if all works correctly , M3 S555 should give me nicely controlled 555 rpm.

There is no sensor i/p on GRBL to monitor spindle speed but this kind of controller should ensure we do "know" the actual spindle speed, once GRBL is configured.

Anyway, thanks again for pointing out where the S goes. That is the point I was getting stuck on. It seems that G97 I found may be specifically for lathe chuck speed. Since complex machines have both milling tools and lathe chucks , I guess the standard treats them differently. Though this does not  seem to be reported clearly on the many results I got.

chamnit, 2019-04-17:
The best way to “know” your spindle speed is to have a spindle controller that can convert the voltage output from Grbl and have a closed loop control system to keep the speed there unloaded and under load. The problem is that most users don’t have access to or is too expensive for specialized electronics like this. It’s easy to blame Grbl for its limits but this isn’t entirely one of them. 

But as a way to help users, I did install a nonlinear spindle speed output to help with fitting Grbls output voltage to actual speeds measured. There is a small section in the config.h file that tells you about it. The process is somewhat involved and uses a piecewise linear curve fit solver in Python to determine the constants to input. This only helps in the unloaded condition. There is no feedback loop to control the speed when the spindle is loaded but at least it’ll get you closer.

J-Dunn, 2019-04-17:
I agree. Closed loop f/b on spindle speed seems a little OOT when you are relying on open loop stepper motors to control position.   The current set up is totally in line with the intentionally simple design philosophy of GRBL.  Thanks for the heads up on the non linear control. I had not seen that. What version was that introduced in?

dommaguyver, 2022-11-28:
Hello everyone.  Thanks for all the help you are all providing.  I just setup my Laser/CNC Router with 3018 control board and this is what I am showing. I did this to show how the spindle value and settings relate, in Candle software anyway.  I put a tachometer and volt meter to my spindle shaft and motor and progressively increased the value in the "Spindle Speed" setting by a value of 100 starting at 100 and went to 1000.  This is what I got.

Speed Setting      Volts     Actual measured RPM
-----------------------------------------------------  
100                     1.7         713
200                      4.9        2170
300                      8           3500
400                     10.8       4780
500                     13.3       5890
600                     15.3       6870
700                      17.5       7835
800                      19.6       8810
900                      22.12     9950
1000                    23.3      10530  

My GRBL $30 value is set to 1000, so any attempt to do tests using speed value over 1000 made no difference.
I tried this 3 times and each time got the same results, so I am confident that they are good.
Hope this helps.
Dom

dommaguyver, 2022-11-28:
Shoot.  Sorry for crappy table above showing the different values.  Basically the speed values are on the left 100 to 1000 and the volts is 1 space over to the right and then the RPM 1 more space over to the right after the volts.

dommaguyver, 2022-11-28:
> Hello,
> Try to set $30=10000 or $30=10530 for your router and adjust the value in
> Candle settings.

Hi.  Ok I tried what you asked but it seems that it will not change the
value in the file.  Although it does not give me an error and does say
$30=10530 < ok, it does not actually change it.  I tried other values like
1001 and it seems that anything above 1000 will not take.

mar0x, 2022-11-28:
So, when you issue `$30=10000` and then `$$` it still shows `$30=1000`. Is it true?

dommaguyver, 2022-11-29:
Yes, that is correct.

J-Dunn, 2023-03-23:
GRBL is accepting the command and clipping the value according to $30.  That is why it returns "ok".  Because it is OK.

01Vineet, 2024-04-22:
my servo motor is not working with the grbl plotter please help me with this its not running with g codes

aimeiz, 2024-06-02:
Hi. 
I'd lite to set real spindle RPM by M3S command instead of percentage.
My spindle max rpm is 16000. I tried to set it by $30 parameter, but regardless what i set by $30, SPINDLE rpm is controlled by M3S command using percentage multiplied by 10, so M3S1000 is max speed, M3S500 is 50%, S0 is 0.
What and where I could set in firmware source code to make real RPM control?