gnea/grbl #638: Unexpected error was detected error:22

JJZ71, 2019-04-19:
[image]

[image]

Hi, all I have a brand new UNO R3 and CNC Shield. 

I am running the same GCODE multiple times and it fails 3/4 of the time with this error but at different points in the code.

It fails even if I run it $C

I have tried CNCjs and UGS , 2 different USB cables and 2 different notebooks.

Can anyone suggest a fix?

MeJasonT, 2019-04-19:
You appear not to have set a feedrate

send F100 for example 

take a look at 
[https://github.com/gnea/grbl/issues/167](url)
[https://github.com/gnea/grbl/issues/255](url)

JJZ71, 2019-04-19:
First line of the Gcode is: 

G1 X1 Y1 F180

If the feedrate was missing would it run successfully 1/3 of the time?

MeJasonT, 2019-04-19:
ok got reply, having a look now

MeJasonT, 2019-04-19:
Just for reference
This is a link to a useful simulator to test code
[https://camotics.org/download.html](url)

JJZ71, 2019-04-19:
Thanks MeJasonT, I installed and ran the gcode, all I could see was one warning about not having a tool selected.

MeJasonT, 2019-04-19:
code ran fine using grbl-panel
I have a uno attached to pc without machine connected running version 1.1f Grbl  as a test bed.
It looks like your problem may be to do with rf/electrical noise which is affecting your communications.
Try running the same test i have without your machine connected to the Uno R3 and see if that's ok - then reconnect your machine and try.

Camotics is good for checking out paths but isn't machine specific to the grbl fur-balls may not necessarily show up

JJZ71, 2019-04-19:
MeJasonT,

I tried running it in check code mode $C and it still failed.

Can you tell me how to run the test on the UNO without the machine connected?

MeJasonT, 2019-04-19:
If you have a spare Arduino Uno then simply upload grbl to it and traet it as if the machine was connected to it, otherwise - is the Uno R3 inserted into your control board or is it micro processor built into your board.
you want this on its own - stops you from destroying a perfectly good machine whilst starting out.
[image]

is this your control board
[image]

MeJasonT, 2019-04-19:
you should be able to then run code live or in check mode.  As grbl does not have axis feedback and relies on its own estimation of machine position, it runs quite happily without a machine connected.

JJZ71, 2019-04-19:
I have an UNO R3, is sounds like it might be rf/electrical noise

MeJasonT, 2019-04-19:
Ok check list time
Screened cables for motors and end stops, preferably wired separately and earthed at one end (controller end making it the closest point to earth potential)
- not machining 0v/gnd/common but an external earth into a metal ground stake if possible
- try not to use electrical outlet earths for grounding the machine
Add ferites to motor and signal wires
perhaps look at getting a mains filter and an opto isolated USB cable/interface
connect any metal machine parts to earth (if its a metal framed machine).
look at rf as being like the old Frankenstein films where you could see lightening travelling around the room, that kind of like what rf does but invisible - loves metal as it acts as a huge aerial.

MeJasonT, 2019-04-19:
If you disable hardware end stops and homing.  set your machine up to a safe 0,0 home position
using software limits you may discover that the noisy coms issue is with the end stops.  It is possible to use the machine without hardware limits and without homing - you just have to be careful how your machine behaves

JJZ71, 2019-04-19:
Thanks, MeJasonT, I'll play with it tomorrow and report back.

chamnit, 2019-04-19:
This is an EEPROM issue. See the FAQ. Try not to stream three consecutive g10 commands. You can combine the into one.

JJZ71, 2019-04-19:
Chamnit, so your saying I need to run a $RST=*

109JB, 2019-04-19:
Both of the pictures in the first post show a reset occurred somehow. This first one shows that after the G10 commands a portion of a parser state response message was received ( [GC:G1 ), and then the reset message ( Grbl 1.1g ['$' for help] )

Once the reset occurs, any programmed feedrate is cleared.

[image]

Likewise, in the second one you can see a partial status report message and a reset message. Again, the reset is going to clear any feedrate info.

[image]

I can't say why there is a reset occurring, but it looks like whatever is causing the reset is what is causing your problem.

cheton, 2019-04-20:
The character-couting protocol is used in CNCjs to stream a G-code program to Grbl, putting "G10 L20" in the middle of a G-code program will possibly cause a reset because it will persist work coordinate settings to EEPROM, and the sender have to wait for the queue to finish.

https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#streaming-protocol-character-counting-recommended-with-reservations

> When Grbl stores data to EEPROM, the AVR requires all interrupts to be disabled during this write process, including the serial RX ISR. This means that if a g-code or Grbl $ command writes to EEPROM, the data sent during the write may be lost. This is usually rare and typically occurs when streaming a G10 command inappropriately inside a program. For robustness, GUIs should track and detect these EEPROM write commands and handle them appropriately by waiting for the queue to finish executing before sending more data. Note that the simple send-response protocol doesn't not suffer from this issue.

@JJZ71 You can try appending a `%wait` command to each "G10 L20" command to wait for the planner to empty. `%wait` is a special command in CNCjs that will wait until all "ok" responses to be returned.

109JB, 2019-04-20:
>The character-couting protocol is used in CNCjs to stream a G-code program to Grbl, putting "G10 L20" in the middle of a G-code program will possibly cause a reset because it will persist work coordinate settings to EEPROM, and the sender have to wait for the queue to finish.

Except in his second example screen there is no G10

cri-s, 2019-04-20:
I just have submitted PR #641 for eeprom prblem. This should fix the issue.

109JB, 2019-04-20:
>I just have submitted PR #641 for eeprom prblem. This should fix the issue.

It may fix an issue, but I don't think it is going to fix the OP's issue.  If you look at the second screen, there were normal G1 commands preceding a reset. There weren't any commands that would be accessing/writing to EEPROM. Am I missing something?

[image]

JJZ71, 2019-04-20:
I changed from CNCjs to UGS and enabled Single Step Mode and status polling the error 22 have almost completely stopped.  Hopefully, it will help narrow down the issue a little further.

[image]

MeJasonT, 2019-04-21:
> This is an EEPROM issue. See the FAQ. Try not to stream three consecutive g10 commands. You can combine the into one.

I'm following the thread with interest
i'm trying to figure out what indicated it was an EEPROM error, from the code example i could see how the feedrate was missing but knowing once its been sent will remain in grbl until its reset or the command is overwritten.  That's what gave me the impression it only happened every so often and we don't know from the example if the board was reset between attempts.  Would it be possible for someone to direct me to the FAQ that Sonny has mentioned.