gnea/grbl #640: Is there a way to run a solenoid Z axis?

davidelang, 2019-04-19:
I have a PCB mill that I'm trying to retrofit to run from grbl (cnc shield). It doesn't have a motorized Z axis, the Z axis is spring loaded with a solenoid that drives it down against a stop when triggered.

Is there a way to run this Z with grbl? or would I have to put a more normal Z axis on the machine to use it?

cri-s, 2019-04-19:
Use the dir signal for the solenoid.
If you need add safety for certain programs, sometime its necessary, use m3
output as enable. Build the and circuit using two resistors as example for
driving the uln driver, especially if you need chopping the solenoid using
pwm.

davidelang, 2019-04-19:
The signal to the solenoid needs to be on for the entire time the Z axis needs 
to be down, will the dir signal will only be on only during the move? or will it 
remain on until there is motion in the other direction?

If it stays on, is this something that can be made part of the defined behavior 
so that it doesn't change in the future?

David Lang

cri-s, 2019-04-19:
The dir signal get its default state on reset, inverted or not and then it
changes as needed on move and stay in that state. As the solenoid need 10ms
to go down you must calc acceleration and travel distance in order that the
axis don't move before z is effectly down or up.

MarkJB, 2020-05-09:
I've just tried this for a grbl powered drawing machine and a solenoid driver board I built (https://github.com/MarkJB/drv120_solenoid_driver_module) and I can control the solenoid with `G0Z-1F100` (down) and `G0Z1F100` (up) but when I issue a G0 or G1 X10Y10F100 the solenoid always lifts. It appears the dir pin is always reset on move.

Any ideas why?

MarkJB, 2020-05-13:
I've hooked up a scope to both Z_DIR and Z_STEP and can see the Z_DIR pin change state when I issue a move in X and/or Y. I'm not changing anything about Z so can't see why Z_DIR is changing state.

I've been trying to track down why this is happening in the code, but not having much luck. I thought it might be something to do with ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING but disabling it hasn't helped.

langwadt, 2020-05-13:
I think the simplest hack would be to add some code in stepper.c around line 324 to only change Z direction if the Z step bit is set

Aggebitter, 2020-05-26:
Why not use flood or spindel enable pin ? 
Add it to your G-code by replace Z- with spindel enable m3 s1000 and m3 s0 for positive values as a example

MarkJB, 2020-05-26:
@langwadt I couldn't see how to implement that. 

I think the Z_DIR change is coming from the planner. Maybe...

@Aggebitter Spindle pin was plan b.
As for why use Z_DIR? Mainly because I didn't want to have to modify the various programs that generate the gcode for drawing machines (or post process the gcode), when it seemed, on the face of it, that using the Z_DIR was an easy option.

langwadt, 2020-05-26:
doesn't matter where it comes from, the z-direction pin gets set or cleared with all the other direction pins in line 324 of stepper.c 

`DIRECTION_PORT = (DIRECTION_PORT & ~DIRECTION_MASK) | (st.dir_outbits & DIRECTION_MASK);`

replace it with some thing like:

`DIRECTION_PORT = (DIRECTION_PORT & ~((1<<X_DIRECTION_BIT)|(1<<Y_DIRECTION_BIT))) | (st.dir_outbits & ((1<<X_DIRECTION_BIT)|(1<<Y_DIRECTION_BIT)));`

`if(st.step_outbits & (1<<Z_STEP_BIT)) 
	DIRECTION_PORT = (DIRECTION_PORT & ~((1<<Z_DIRECTION_BIT))) | (st.dir_outbits & ((1<<Z_DIRECTION_BIT)));`

MarkJB, 2020-05-29:
@langwadt thanks! That seems to be working.

ygjfggfgfj, 2020-08-25:
hello there
i dont have any programming knowledge 
could any one help me how to set stepper motor as spindle in GRBL library?