gnea/grbl #642: How to send Ascii character to trigger safety door operation

MeJasonT, 2019-04-20:
I'm trying to find out how to send " ˶  " character to activate safety door operation.
If for example i was to write a tool change function into my post processor which when it received the tool identifier T1 it would run ascii commands.  It accepts a feedhold if i get it to send its character but nothing happens sending " ˶  " character 0x84.

I have enabled the safe door operation in config.h but its always possible i've forgotten to do something else?

I'm wanting to trigger as part of the gcode file, not the physical pin input

109JB, 2019-04-20:
0x84 is not a printable text character therefore you can't just add it to the g-code.

You could edit the code to change the character used for safety door activation and change it to a printable character to accomplish what you want. edits would be needed in config.h and serial.c

MeJasonT, 2019-04-21:
Hi John,
looks like "¬"  is in the extended ascii table,  I take it that the change made to serial.c will be to accept character which is in either the low 128 bit characters or or above 128 bit character table using the switch case statement, is this what you meant - what is the change i would need to make.  In any case  using "¬" didn't work.  I ran a file on candle and sent the character, well it just ignored it did it.

109JB, 2019-04-21:
for what you are trying to do I would just use a character that you can type from the keyboard. There are plenty that Grbl compliant g-code doesn't use.  For example to change the code to use "&" to trigger the door you would do the following.

First, in config.h you need to change the following line for the new character

#define CMD_SAFETY_DOOR 0x84    --->   #define CMD_SAFETY_DOOR '&'

then in serial.c, about line 158 you will see the following line

case CMD_SAFETY_DOOR:   system_set_exec_state_flag(EXEC_SAFETY_DOOR); break; // Set as true

just move this up about 3 lines to where you see the line starting with 
case CMD_FEED_HOLD: 

 That's all.  Now a "&" in the stream should cause the safety door to trigger.

MeJasonT, 2019-04-21:
Thanks i will take a look at it tonight, i'm currently up to my neck in wiring.
Im converting my suds pump from 3 phase to 24v by replacing the motor and removing the old machine wiring putting in a single phase 24 vdc transformer instead. wiring the door switch and connection for a probe.

you know how when something works you try everything to avoid opening up the controller panel and replacing the IC with one that's been reflashed - dragons be down there and all that

MeJasonT, 2019-04-21:
Thanks John,
its looking good - so far.
I'm using an Atmega 328 micro.  I have done a little searching but not found any conclusive information regarding the spindle retract and resume, just a note on one issue where sonny is saying something about there is no write-up in a wiki yet as the feature is just for OEMs.  I figure i'm supposed to enable parking is that correct.  Also on resume the coolant isn't reactivated, i can remember reading about this issue a while ago but can seem to find it again.  its probably on the old grbl site.  Almost got a fully functioning DIY VMC.  Another issue im having is Grbl-Panel locks the command input box whilst its running a file, so cant send '&' , if its activated in code the monitor windows stops reporting after resume.  Candle CNC has no clue what to do when the safety door is activated, luckily grbl does its thing and waits for a resume - candle does come back to life when operation is resumed.  Just found some code today which automatically hunts for a work piece's origin and sets the G54 WCS, why can i have that in grbl-panel.  I must figure out how to get visual studio to stop giving me errors and warnings every time i try to open and compile Gerrits code.

MeJasonT, 2019-04-30:
Thanks again for your help John.
Managed to run several code blocks now with '&' buried in the code and the machine quite happily reads it and performs a retraction, spindle-coolant stop and pause.
finally fitted door switch and all works well.

closed 2019-04-30 by MeJasonT

Hubert880411, 2020-09-13:
> Thanks again for your help John.
> Managed to run several code blocks now with '&' buried in the code and the machine quite happily reads it and performs a retraction, spindle-coolant stop and pause.
> finally fitted door switch and all works well.

Hi,Jason.Can you explain how to do that in detail?I still can't understand how you solve the problem.I did just like John said,but grbl safety door still can't work.

MeJasonT, 2020-09-13:
Have you enabled the safety door function in the firmware config.h

MeJasonT, 2020-09-13:
This was my config.h
I say was as im now using Tormach's PathPilot

`/*
  config.h - compile time configuration
  Part of Grbl

  Copyright (c) 2012-2016 Sungeun K. Jeon for Gnea Research LLC
  Copyright (c) 2009-2011 Simen Svale Skogsrud

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

// This file contains compile-time configurations for Grbl's internal system. For the most part,
// users will not need to directly modify these, but they are here for specific needs, i.e.
// performance tuning or adjusting to non-typical machines.

// IMPORTANT: Any changes here requires a full re-compiling of the source code to propagate them.

#ifndef config_h
#define config_h
#include "grbl.h" // For Arduino IDE compatibility.

// Define CPU pin map and default settings.
// NOTE: OEMs can avoid the need to maintain/update the defaults.h and cpu_map.h files and use only
// one configuration file by placing their specific defaults and pin map at the bottom of this file.
// If doing so, simply comment out these two defines and see instructions below.
#define DEFAULTS_GENERIC
#define CPU_MAP_ATMEGA328P // Arduino Uno CPU

// Serial baud rate
// #define BAUD_RATE 230400
#define BAUD_RATE 115200

// Define realtime command special characters. These characters are 'picked-off' directly from the
// serial read data stream and are not passed to the grbl line execution parser. Select characters
// that do not and must not exist in the streamed g-code program. ASCII control characters may be
// used, if they are available per user setup. Also, extended ASCII codes (>127), which are never in
// g-code programs, maybe selected for interface programs.
// NOTE: If changed, manually update help message in report.c.

#define CMD_RESET 0x18 // ctrl-x.
#define CMD_STATUS_REPORT '?'
#define CMD_CYCLE_START '~'
#define CMD_FEED_HOLD '!'

// NOTE: All override realtime commands must be in the extended ASCII character set, starting
// at character value 128 (0x80) and up to 255 (0xFF). If the normal set of realtime commands,
// such as status reports, feed hold, reset, and cycle start, are moved to the extended set
// space, serial.c's RX ISR will need to be modified to accomodate the change.
// #define CMD_RESET 0x80
// #define CMD_STATUS_REPORT 0x81
// #define CMD_CYCLE_START 0x82
// #define CMD_FEED_HOLD 0x83
#define CMD_SAFETY_DOOR '&'
#define CMD_JOG_CANCEL  0x85
#define CMD_DEBUG_REPORT 0x86 // Only when DEBUG enabled, sends debug report in '{}' braces.
#define CMD_FEED_OVR_RESET 0x90         // Restores feed override value to 100%.
#define CMD_FEED_OVR_COARSE_PLUS 0x91
[... 533 lines cut ...]
#define RPM_POINT23  9627.8  // Used N_PIECES >=3. Junction point between lines 2 and 3.
#define RPM_POINT34  10813.9 // Used N_PIECES = 4. Junction point between lines 3 and 4.
#define RPM_LINE_A1  3.197101e-03  // Used N_PIECES >=1. A and B constants of line 1.
#define RPM_LINE_B1  -3.526076e-1
#define RPM_LINE_A2  1.722950e-2   // Used N_PIECES >=2. A and B constants of line 2.
#define RPM_LINE_B2  8.588176e+01
#define RPM_LINE_A3  5.901518e-02  // Used N_PIECES >=3. A and B constants of line 3.
#define RPM_LINE_B3  4.881851e+02
#define RPM_LINE_A4  1.203413e-01  // Used N_PIECES = 4. A and B constants of line 4.
#define RPM_LINE_B4  1.151360e+03

/* ---------------------------------------------------------------------------------------
   OEM Single File Configuration Option

   Instructions: Paste the cpu_map and default setting definitions below without an enclosing
   #ifdef. Comment out the CPU_MAP_xxx and DEFAULT_xxx defines at the top of this file, and
   the compiler will ignore the contents of defaults.h and cpu_map.h and use the definitions
   below.
*/

// Paste CPU_MAP definitions here.

// Paste default settings definitions here.

#endif`

Hubert880411, 2020-09-13:
> This was my config.h
> I say was as im now using Tormach's PathPilot
>
> `/*
> config.h - compile time configuration
> Part of Grbl
>
> Copyright (c) 2012-2016 Sungeun K. Jeon for Gnea Research LLC
> Copyright [...]

Yes,I changed.And I use Arduino to send g code and type '&' on Arduino serial monitor when g code is running.By the way,I use Arduino mega1 send g code and real time command to Arduino mega2(grblupload is at here).Is that work?Maybe you can tell me other ways to send grbl real time command.That will be helpful.Thanks Jason.

MeJasonT, 2020-09-13:
Is your safe door function working when you open the door switch (wired into the pause pin - I think it was)
if you use a text editor you can write manual gcode and include the character that you are using to programaticly trigger the door function.  compare your config with my one above and see what changes there are.

Hubert880411, 2020-09-14:
> Is your safe door function working when you open the door switch (wired into the pause pin - I think it was)
> if you use a text editor you can write manual gcode and include the character that you [...]
Got it.I will try.Thanks for your help,Jason.

Hubert880411, 2020-09-14:
> Is your safe door function working when you open the door switch (wired into the pause pin - I think it was)
> if you use a text editor you can write manual gcode and include the character that you [...]

Hi,Jason another question. I also want to execute the "0x92" real time command.Is the same way just like safety door to do that?

MeJasonT, 2020-09-14:
remind me the character is ( ' ) right (ignore brackets, tried quotes it looked daft)
just send the character in your gcode.
grbl has a set of predefined characters eg ? to report its position etc. it parses all incomming messages and if it sees its own communication characters it responds to them.  One problem you may have is the buffer.  Lines of code are sent to grbl which will accept a couple of line of gcode to be stored and then read them off the buffer in order.  However the direct grbl commands are acted upon straight away so you may find that some of the gcode isnt finished when you send the door open command etc  it became a headache for me so i gve up and didnt bother sending commands in the end.   Even Pathpilot has its anal issues.  if i dont shut the doors i cant home the machine - once homed i can eg to set up work, however it works fine if i open the door during operation apart from it doesnt retract.  so you see some things you end up having to live with.  Its brilliant what has been acheived with grbl, getting all that cod on a Atmega 8 bit micro and it still runs a cnc machine to a reasonable level is Tops in my book.

MeJasonT, 2020-09-14:
i never switched to the mega

Hubert880411, 2020-09-14:
Thanks for sloving my question.I appreciate Jason.