gnea/grbl #644: Problem with homing sycle

samehherit, 2019-04-22:
Hi every body 
Im using grbl 1.1f , i faced a problem when im going to home sycle 
When i press home the z axis move normally 
The problem is when the xy axis move to complete the sycle it het the switches without stopping   
I dont knew where the the problem is ??

My limit switches is ( NC )

My grbl settings:

$0=10
$1=25
$2=0
$3=7
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=50
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=20.000
$101=20.000
$102=445.000
$110=2000.000
$111=2000.000
$112=300.000
$120=300.000
$121=100.000
$122=10.000
$130=3000.000
$131=3000.000
$132=100.000

samehherit, 2019-04-23:
Any body help ??

109JB, 2019-04-23:
Is the activation of the limit switches showing up in the status reports?  You should see a PN: X if the x limit switch is activated.  You can check this without the machine moving by activating the switch manually.

samehherit, 2019-04-24:
Thanx for replying
I tested my switches triggering its showing these messages / 

[verbose]<alarm|mpos:0.000.0.000.0.000|f:0|Pn:x>

[verbose]<alarm|mpos:0.000.0.000.0.000|f:0|Pn:y>

[verbose]<alarm|mpos:0.000.0.000.0.000|f:0|Pn:z>

But the problem is still on

109JB, 2019-04-24:
You may have the homing direction for XY reversed. You can just try $23=0.

samehherit, 2019-04-28:
The same thing nothing deferent
Thanks a lot for your help

samehherit, 2019-04-28:
Hi friend 
Finally my problem were fixed and the machine homing very well 
I discovered where the problem was
The terminals  of x and y cable were in a reversed position on the arduino shield pins

Thank you very much for helping

closed 2019-05-06 by samehherit