gnea/grbl #645: CAMTool v3.3 Optical Limit Switches CNC 3018 Pro

hydra, 2019-04-23:
Since this is where I ended up after getting a CNC 3018 Pro I thought I'd share some images of the limit switches I used on my machine.

These are the ones I bought.

[image]

They are available at many online retailers such as [Amazon](https://www.amazon.es/gp/product/B07DK3J88S/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1), [AliExpress](https://www.aliexpress.com/item/6pcs-lot-Optical-Endstop-Light-Control-Limit-Optical-Switch-for-3D-Printers-RAMPS-1-4/32861519345.html?spm=2114.search0104.3.57.7345289eQr1uqK&ws_ab_test=searchweb0_0,searchweb201602_5_10065_10068_319_10891_317_10548_10696_10084_453_454_10083_10618_10304_10307_10820_10821_10301_537_536_10059_10884_10887_321_322_10103,searchweb201603_56,ppcSwitch_0&algo_expid=562fa36a-b63c-4678-9a5c-5365b982019b-8&algo_pvid=562fa36a-b63c-4678-9a5c-5365b982019b&transAbTest=ae803_4)

This is how they were installed.

[image]
[image]
[image]
[image]

Here is how they were connected.

[image]

The optical sensors have `V` `S` and `G`.

* Connect 'G' to the bottom row of the header, for the corresponding axis.
* Connect 'S' to the top row of the header, for the corresponding axis.
* Connect 'V' to the 5v, top left of the header.

Since there is only one 5v output, I used some header pins and a cable to make a simple adapter to allow all the 5v optical sensor power connections to be connected to each other.

Sensors were positioned so they they are cannot be touched by any moving part.

For the white plastic part that goes into the optical sensor I simply cut up a plastic container and bent it to shape. Then used very stick double-sided foam tape to stick them to the following locations.

x - bottom of the head.
y - the underside of the bed, right hand side.
z - side of the head.

here's the important parts of the machine state after using UGS Platform v2 nightly.

```
$5 = 1    (Invert limit pins, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 2    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$130 = 290.000    (X-axis maximum travel, millimeters)
$131 = 180.000    (Y-axis maximum travel, millimeters)
$132 = 29.000    (Z-axis maximum travel, millimeters)
```

When these optical sensor are not blocked by an object the red LED on them is bright.  This is handy so that you know they have power.  When the optical sensor is blocked the red LED on them goes out or goes dim.

Using black plastic would likely be better than white, but it was all I had.

I created a google photos shared album with the images, here - https://photos.app.goo.gl/rQoACTyaKqubXJK3A

Hope this helps people.

Note: I tried to add this to the wiki page but the wiki doesn't allow image uploaded but issues do.

samehherit, 2019-04-23:
Good job thanx alot

MechaSteve, 2019-04-24:
A minor suggestion:
These type of sensors are most accurate when the flag (the piece of plastic blocking the LED) moves into the slot parallel to the PCB. 

How you have the Y and Z axis mounted is perfect for this, but the X axis may not be as accurate as possible.

This more ideal mounting also has the advantage of not pressing against the sensor in the event of an over-travel / disconnected wire, etc.

hydra, 2019-04-27:
> ... not pressing against the sensor in the event of an over-travel / disconnected wire, etc.

I placed the sensors so they cannot be squashed by over-travel.  other things make contact first on all three axes.  I paid particular attention to this when mounting them.

Others reading this should also pay attention to this good point.

lexelby, 2019-05-02:
Awesome, thank you!

It looks like you have just one switch per axis, right?  Do I understand correctly that that is only useful for homing?  If I want to protect against traveling too far in both directions for an axis, do I just connect the two switches to the same header pins in parallel?

hydra, 2019-05-03:
@lexelby hard limits for homing and protection in one direction. then enable soft limits for maximum travel in the other direction.

it works well for me.

lexelby, 2019-05-03:
Oh, of course!  Why didn't I think of that.  Thanks :)

kenmaistre, 2019-05-24:
Or you can use 2 switches per axis connected in parallel. I did mine this way just to be safe. Soft limits work great as long as there are no software issues. Its easy to forget which unit of measure you are in, after all 25 mm is not the same as 25 inches...

rymrks, 2019-06-08:
> Or you can use 2 switches per axis connected in parallel. I did mine this way just to be safe. Soft limits work great as long as there are no software issues. Its easy to forget which unit of measure you are in, after all 25 mm is not the same as 25 inches...

This is exactly what I am trying to achieve but my knowledge in this field is as a novice. I would really appreciate if you could help me by explaining in detail how you wired up your switches in parallel. Thank you in anticipation. Roy

kenmaistre, 2019-06-22:
> This is exactly what I am trying to achieve but my knowledge in this field is as a novice. I would really appreciate if you could help me by explaining in detail how you wired up your switches in [...]

Here is a link on how to wire limit switches. I hope this helps:
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

rymrks, 2019-06-22:
That is indeed a great help to me, thank you for replying with this information I am very grateful to you.

Regards Roy

kenmaistre, 2019-06-25:
> That is indeed a great help to me, thank you for replying with this information I am very grateful to you.
> 
> Regards Roy

Certainly, sorry it took a while, i don't check those messages often. To make your life easier stick with the same kind of switches as shown, you can find micro-switches on Amazon at 20-25 pieces for about $6-7. Remember you will only use 2 pins off the switch, in the Normal Open (NO) mode. Good luck!

lexelby, 2019-06-29:
Careful, folks.  The Z axis voltage regulator on my camtool v3.3 board just caught fire and let out the magic smoke.  Time for a new grbl board...

rymrks, 2019-07-09:
I have fitted the limit switches as suggested by kenmaistre and happy to report all works as it should with one slight exception. That is, when I hit the microswitch limit I then have to, reset the software, then press unlock within UGS (as expected), but not once but twice before the travel commands will work again. After the first reset it moves 0.5mm away from the limit before tripping the switch again. I don't understand why this would be, can anyone shed some light on this problem? When I used the optical limit switch setup as shown by hydra then the limit switch behaves as it should, ie, reset and unlock and the travel command works right away.

jongardneruk, 2019-07-13:
Thanks for the share - the photos really helped when I was doing something similar on my own setup.

buttim, 2019-09-21:
Can you please tell me if your switches give a high or low output signal when unobstructed?
I'm trying to install [similar switches](https://www.banggood.com/3pcs-Optical-Endstop-Limit-Switch-Sensor-with-1M-3Pin-Cable-for-3D-Printer-p-1399210.html) but I'm having a hard time. Mines give a low output when unobstructed (and the led is off too)

rymrks, 2019-09-21:
Well I removed my Optical Limit Switches in favour of micro switches. The reason was I could only get them to work in the plus axis directions only, due to the limitations on my control board. As I remember my OLS were giving a high output with the led on when unobstructed . Hope this helps.

EBEALIEN, 2019-10-30:
Goodmorning everyone,
forgive my intrusion, if I wanted to insert 6 xyz micro switches, how do I connect them ?? each axis in parallel (2) on the zyx base?
Thank you very much

mzveg, 2019-12-05:
Hello to everyone!
I would like to connect my 5 optical switches (identical to those used by hydra) to my Camtool V3.3 controller board (seems to be the same than hydra too...) .
The photograph is perfect to understand the connections for GND and 5V (I'll need a 6 pins DuPont connector or the 5Vcc).
I still have a doubt on wiring the signal pins in parallel on the same header pin (2 for the X pin and 2 for Y pin, but only one for Z pin).
I confirm that the signal pin is normally low (unobstructed) and goes high when obstructed. The led works on the opposite.
Could somebody help me on this topic ?
Thanks in advance !!!
PS: Is there a place where I could find a description of the  pinouts of the different headers or connectors of this board ?

hydra, 2019-12-06:
Hi @mzveg 

The `$5 = 1    (Invert limit pins, boolean)` inverts the input.

The optical sensors have 3 connections, V G and S,  V is supply voltage, G is Ground and S is Switch output (inverted).  As I noted the header pin has multiple G an and S connections but only a single V connection so you have to connect a single V pin from the PCB to each V pin in the switches by using some breakout board/headers like the one I made.

The wiring/photos I details are definately correct for all 3 axis, if you wire it and configure it the same as me you should have no problem.

bdurbrow, 2019-12-06:
Note that running these in parallel only works because of the type of output the TCST2103 has; other "logic" or "push-pull driver" output devices will fight if connected in parallel.

mzveg, 2019-12-06:
Thanks to @hydra & @bdurbrow  for their support !
I'll try to do some tests during the week-end ...

mzveg, 2019-12-19:
Hello !
I had time to run some tests on my TCST2103 limit switches ...
I wired two of them in parallel (common Vcc, common gnd and common "Signal") and there is a problem : the "common" output always remains at low level and the two leds always remain lit when I put an obstacle in one of the detectors...
@bdurbrow : did you test this configuration as mentioned in your recent post ?
Would somebody have experience on this kind of wiring ?
Thanks in advance !
PS : Each detector works fine when the outputs are separate ...

bdurbrow, 2019-12-20:
Doh! Those pull down when not blocked. So wiring in parallel will generate a high output when all of them are blocked... which is the opposite of what you want. Instead, you need an OR gate chip to combine the inputs. Something like a 7402 on a piece of perfboard; and as that’s actually a NOR gate, you can either configure Grbl to invert the input or take the output from one gate and run it through another to invert it again (and just connect the unused input on the second NOR gate to ground).

mzveg, 2019-12-20:
@bdurbrow  : I'm puzzled ... Two weeks ago you wrote that wiring two sensors in parallel was fine because they were based on a TCST2103  switch. and yesterday, you wrote that extra circuitery is needed because of pull downs (which seem in fact to be pull-ups) . 

Is there anybody having actually tested parallel hookup of two TCST2013 based sensors on a CAMTool v3.3 board ?

bdurbrow, 2019-12-21:
Sorry about the confusion... I'll try to clarify.

The original comment I made was only partially correct: it "works" in the sense that it doesn't cause a short circuit and possibly destroy your hardware. Connecting two "totem-pole" style drivers together can do that when one driver is trying to pull up to the positive voltage supply, and the other is trying to pull down to ground.

However, in this case the specific device in question - the TCST2103, at least as configured on the version of the modules that I have - can only pull down to ground. It depends on something external to pull the voltage on the signal line up to the "high" level (typically +5v). On the common modules used for 3D printers, that something is a resistor connected between the V and S pins on the module; when the TCST2103 goes active, it overpowers the resistor, and the voltage on the S pin is pulled very close to ground (my measurements just now on a breadboard show about 0.5v as the low level that it pulls to). The resistor is sized such that connecting it directly between the V and G pins (and thus applying the full power supply voltage across it) won't be a problem.

Connecting two modules in parallel "works" in that when neither TCST2103 is active, the S line is pulled up by the resistors in both modules, resulting in a high output state. When one or both TCST2103 units goes active, it pulls the S pin down to ground overpowering both resistors - but the TCST2103 units themselves don't fight each other; and there's never a case where there is a dead short between the positive power supply and ground (the resistors are always in the way).

The problem is that the TCST2103 is active when it's beam is _not_ being blocked: just sitting there, it's always pulling the S line to ground. This results in the opposite logic to what you want.

Direct Parallel Connection:

```
Beam Blocked: 1 | 2 | Output:
-----------------------------
              N | N |  Low
              Y | N |  Low
              N | Y |  Low
              Y | Y |  High              
```

Desired Logic:

```
Beam Blocked: 1 | 2 | Output:
-----------------------------
              N | N |  Low
              Y | N |  High
              N | Y |  High
              Y | Y |  High
```

The only way to get that logic function with these sensors is to put an OR gate in between the sensors and the single input pin on the Grbl card.

I don't have a CAMTool v3.3 board; the CNCs that I'm working on are all based on the Arduino Mega 2560; and run a modified and extended version of Grbl-Mega. So, I've (obviously) not tested the hardware with that board; I'm just working off of the TCST2103's datasheet and a module I have here in a breadboard on my bench for software development testing.

P.s. - it occurs to me that perhaps the "slang" I used in my previous post might not be clear to an international audience. "Doh!" is from a popular cartoon here; it is said with a grimace, and basically means "Oops" or "I made an error".

mzveg, 2019-12-21:
@bdurbrow 
Many thanks, Britt for the explanation and the time spent to clarify !
No problems with your "slang".
I'll have to test a hookup with OR gates (I should have some of them buried in my workshop, but where ?). 
It also could be done with NOR gates if you change the $5 = 1 setting ...

It's a pity that this CAMTool board has apparently no "official" documentation.
My internet searches did not give any serious info on the manufacturer or designer of this board.

Merry Christmas and happy new year to all !!!

langwadt, 2019-12-21:
two transistors and a resistor would work too, https://imgur.com/cKUIDxZ

MAKINACI72, 2020-05-08:
[image]
Hi,

I made the endstop connection that you made here. But when I made the 5v connection, it stopped working. What could be the reason? If the pin there is burned?

hydra, 2020-05-08:
@MAKINACI72 You probably wired it up wrong, check and double check all your connections!  Do *not* power it up again until you're sure everything is 100% correct.

cliff-pennalligen, 2020-10-08:
thankyou thankyou thankyou.  I have exactly these sensors and a 3018 pro!  It made it easier setting mine up

silviu25, 2022-11-07:
> Can you please tell me if your switches give a high or low output signal when unobstructed? I'm trying to install [similar [...]

Hi, have you found a solution? I have similar switches and struggling with them.....

ncc2707, 2024-02-05:
Thank you soo much! I just got my first 3018 Woodpecker v3.3. Had No "Home" switches at all! Had a few of those Optical Switches in my draw-of-drawers, LOL. Now I'm going to clean the rods, leads, screws, and bearings. I use "Dry Graphite" for all my moving parts. No build up! Next project is a 990 x 990 x 260 mm CNC unit using Nema 34 steppers.