gnea/grbl #648: Question: Homing ends at outer limit of max travel range

Elmi77, 2019-04-27:
Hi,

I have set the maximum travel range to 5000 (directly within defaults.h). Now when I do a homing with X-axis and $23 set to 0, after completion the X-axis ends at position -1.0 (which is correct because pull-off distance is at 1).

But when I do homing with the Y-axis in next step with $23 set to 2 (means inverted movement direction for Y-axis) after homing sequence has completed, this axis is located at -4999.0 - means the pull-off-distance has been calculated from outer limit and not from centre position at 0.0.

What could be the reason for this? What am I doing wrong?

Thanks!

neilferreri, 2019-04-27:
The machine's home position (0,0,0) is located at the positive extent of each axis. If you're moving in the negative direction, you're moving away from that point.

mikemaikaefer, 2019-04-29:
@neilferreri I understand, but this does not explain why it ends up at the outer limit position after homing in negative direction! The position still should be set to something close to 0, not to a limit position!?

chamnit, 2019-04-29:
@neilferreri : Please see the FAQ. There is an explanation there.

chamnit, 2019-04-29:
@neilferreri. Sorry I meant @Elmi77

closed 2019-04-29 by chamnit

Elmi77, 2019-04-30:
@chamnit sorry but there is no explanation in the FAQ! There is an explanation why the coordinates are negative, but I did not find an explanation why they are at the negative **limit** only in case of inverted homing direction.

Elmi77, 2019-04-30:
OK, finally I found it, this statement is responsible for the strange behaviour:

if ( bit_istrue(settings.homing_dir_mask,bit(idx)) ) {
 set_axis_position = lround((settings.max_travel[idx]+settings.homing_pulloff)*settings.steps_per_mm[idx]);

So in case of inverted home-motion the position intentionally is set to the extreme limit by using "settings.max_travel[idx]". Sot sure what the reason is but when I remove these two lines, it works as described.