gnea/grbl #649: Soft limits question

GogoVux, 2019-04-28:
Hi,
im quite new to CNC and apologize for any dumb question.
I hav a little 16010CNC with a woodpecker grbl 0.9 controller board.
Now i have upgraded this mill with endstop switches, and grbl 1.1.
I have set all the $nn for homing and hardlimits - and it it worked ! 
Great ! After homing machine goes to -157, -1, -1. (X-Axis has inverted homing, and a max travel of 158mm, Pull off is 1.0mm).
The following example is only for y-axis, but it is the same for all axes.
The total distance of machine travel in y is ~106mm between limit switches, subtracted twice the savety distance of pull off (1mm) results to 104, which i set for $131.
When i send g0y-104 machine goes to this position, but this is 2mm before endstop. (g0y-104.001 does not work, because of activated soft limits)
When i send g0y0 machine travels to this position - but ends triggering the limit switch.
Same is for g1.  Shouldn't softlimits avoid the triggering of limit switches ? 

N.B. Jog commands are respecting the soft limits.
I'm using Candle as GCode sender.
Is there anything wrong with my setup ?

$0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=1
$24=25.000
$25=500.000
$26=50
$27=1.000
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=5000.000
$111=5000.000
$112=800.000
$120=10.000
$121=10.000
$122=10.000
$130=158.000
$131=104.000
$132=40.000

Thank you in advance

GogoVux, 2019-05-16:
Because there was no reaction, will retry.
My mechanical configuration is for Y axis: Total travel between limitswitches is 106mm.
I want to have a 1mm safety space on both ends before limitswitches are triggered - so i set the software limit $131 to 104.
After homing cycle with 1mm pulloff, machine and work coordinate are -1.
It is possible to reach the machine coordinate position 0 with any G command (G0, G1, ... or Jog) without intervention of GRBL softlimits check - and this triggers the limit switch.
In opposite i can only go to absolute machine position -104 wihout softlimits  intervention.  

Shouldnt be the position after homing **and** pull off set to 0 (instead of -1)  ?
Or in other words: Why is the position 0 assigned to limit switch trigger position, and not to the safe point at pull off position ? 

Setting the 0 at the pulloff position would give the softlimits feature a lot more usefulness ?
How can i setup my system to avoid legal positioning to machine coordinate 0 ?

I would really be happy, if any of the experts could give me a response.

109JB, 2019-05-17:
Uncomment this line in config.h

// #define HOMING_FORCE_SET_ORIGIN // Uncomment to enable.

GogoVux, 2019-05-17:
@109JB ,
thank you very much, this solved (most of) my problem.
Unfortunately has my X axis inverted homing - and the code in HOMING_FORCE_SET_ORIGIN sections 
ignores the inverted limit switch position, and sets the position (FORCE...) to 0 -and not to -Maxtravel.
I changed also the code in HOMING_FORCE_SET_ORIGIN sections of limits.c and system.c, and now i'm happy.
It is now easier for me to calculate positions, and it avoids hitting of limit switches.

Thank you again.

kirsten-dankhil, 2019-08-08:
Hello, 

Assumes the machine has been homed,
// the workspace volume is in all negative space, and the system is in normal operation.

What i should change to work in positive workspace volume with soft limits enabled?

GogoVux, 2019-08-08:
Hello, 
in my opinion you do not need to change anything to have a positive workspace volume. 
After homing, the machine coordinate system is fixed, and checking of softlimits is done by grbl against this coordinate system.
In most (all ?) gcode sender applications for grbl you can set the origins of a working-coordinate systen of your choice.
I'm using Candle. It remembers the offsets between machine coordinates and work coordinates, 
so you can use the buttons "Restore Origin" (and "Safe Position") to move to your desired 0 position.
I'sure that this is also possible with other programs.