gnea/grbl #651: Noisy spindle after upgrading to grbl 1.1

GogoVux, 2019-04-28:
Hi,
after upgrading my woodpecker grbl0.9 controller (it has no version printed on it, i assume version 1) to current grbl1.1  i have noticed a quite loud beeping noise when spindle is on, especially at lower speed values (50/1000 .. 200/1000).
The power of spindle is also reduced.
Because i had changed not only the grbl version (limit switches added, cabling changed, ...),
i have "downgraded" the grbl version back to the original version provided by manufacturer. It reports as grbl0.9j. And the noise was gone (same hardware, different firmware). 
Because the manufacturers firmware hex file is bigger than the original grbl0.9jhex-file, and might have any special modifications, i tried also the official grbl0.9j version.
There is also no beeping noise, and "full" power.
Restoring to 1.1 - and the beep is also restored.
The beeping frequency is around 1 kHz (i'm not a musician), and independent from speed.

Is there any setting or, may be, anything at the config.h to do, to make the spindle speed control 
compatible to the woodpecker board (like the 0.9j)  ?

My current setup for grbl 1.1 is:
$0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=1
$24=25.000
$25=500.000
$26=50
$27=1.000
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=5000.000
$111=5000.000
$112=800.000
$120=10.000
$121=10.000
$122=10.000
$130=158.000
$131=104.000
$132=40.000

The settings for the 0.9 version:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=5 (dir port invert mask:00000101)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=5000.000 (x max rate, mm/min)
$111=5000.000 (y max rate, mm/min)
$112=800.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
 
Thanks for any help.

langwadt, 2019-04-28:
spindle pwm frequency? line 138-143, https://github.com/gnea/grbl/blob/master/grbl/cpu_map.h

GogoVux, 2019-04-28:
@langwadt ,
tried #define at line 141 - and problem is solved.
Thanks a lot for fast response and help !

Remains the question, why was this changed for 1.1 ?  Are there any advantages for other devices, except cheap controllers and spindles ?

Again - Thank You.

closed 2019-04-28 by GogoVux