gnea/grbl #653: Quick clarification required, more machine learning than issue.

MeJasonT, 2019-04-30:
If i run a macro to set a tool height which operates at the X0 Y0 Home position and say that it gets 20mm height (if tool stuck out more than machine 0 full range Bad stuff would happen) It should be possible to set 20mm instead of making the height zero, am i correct.  also if the machine now knows where the bottom of the tool is when i start a job using G54 it will go to those coordinates with the bottom of the tool based on zero part reference.   Its funny but there is loads of information on the web about setting Z0 and tool height but nothing connecting the subject and showing the full operation.
The plan is to do just what i have written to set the tool and then run the job and when the job completes if i have a tool change i will send an & character to park the machine ready for a manual tool change and go through the whole operation again.  It would be nice to move the machine back to 0,0 mid code change the tool, have it re zero then resume back to location after the change.  I know its not possible on the Atmega 328. 
So basically question is i can set tool height in machine coordinates and work coordinates will still  know where to start even if the surface height of the material (in a vice) is at 50mm for example

109JB, 2019-04-30:
Generally one tool is selected as the master tool and it is touched off on the top of the part. Then any other tool is referenced to this tool using a tool length offset (TLO).  The TLO for the master tool is 0.000. If another tool sticks out of the spindle an additional inch (or mm if setting in mm) then the TLO for that tool is 1.000.

The master tool, or TLO=0 can be anything. It can be the spindle nose, the tip of tool 1, the tip of tool 22 or whatever. The important thing is all tool lengths reference from this. I use my probe as my master tool.

Now in Grbl there isn't a tool length table, so the GUI has to do it or you have to do it. Grbl uses dynamic TLO using G43.1. So for example to set a tool length offset of 1.000 you enter the command G43.1 Z1.000

MeJasonT, 2019-05-06:
Thanks John.
I'm working on it, slowly but surely.
I'm looking at making changes to GrblPanel GUI to add additional features, i'm in contact with Gerrit but have not made any significant progress worth reporting.  I'm also trying to create a tool table for cambam - I got tired of asking if some kind sole would write one and there has been no major upgrades or updates for a long time now.  I can see why many users have switched to Fusion 360, just wait until we are all hooked and reliant on it, it wont be presumed free forever.  That's when we will all start heading back to programs like cambam.  I asked cambam for a potential improvement to the program for a facing operation, the reply came from a number of users to write my own.  I didn't consider the response to be in the spirit of the software agreement especially after i purchased my copy.  That's like microsoft asking us to finish their operating system - wait they kind of do in a way its called accepting updates.  I worked out that over the last 10 years i must have received over 4Gb of updates that's larger than the original installation dvd.  Programmers tut.