gnea/grbl #654: Post processor issues

jeffp600, 2019-04-30:
Has anyone been able to use commercial software like artcam or aspire to output usable CNC file ? I get errors on all of them(mach3,gecode,etc.)  Using arduino CNC and gcode sender

zeevy, 2019-05-01:
I don't have any problem with aspire tool path export.
I select: Grbl (mm)(*.gcode)

MeJasonT, 2019-05-01:
Is this barking up the right tree

[https://discuss.inventables.com/t/artcam-2008/58951](url)

[https://forums.autodesk.com/t5/artcam-forum/looking-for-post-processor-in-artcam-to-generate-gcode-for/td-p/6749734](url)

[https://forum.shapeoko.com/viewtopic.php?f=3&t=4125](url)

[http://millrightcnc.proboards.com/thread/873/artcam-save-code-toolpath-grbl](url)

[https://boxzy.dozuki.com/Answers/View/581/How+to+get-make+a+postprocessor+for+ArtCAM.](url)

jeffp600, 2019-05-05:
MeJasonT every click gives not found?????

jeffp600, 2019-05-05:
Artcam 2008 gives that error if u do anything TO pp folder.

schwittlick, 2019-05-05:
Just since I'm around, making it easier to click for you:

https://discuss.inventables.com/t/artcam-2008/58951

https://forums.autodesk.com/t5/artcam-forum/looking-for-post-processor-in-artcam-to-generate-gcode-for/td-p/6749734

https://forum.shapeoko.com/viewtopic.php?f=3&t=4125

http://millrightcnc.proboards.com/thread/873/artcam-save-code-toolpath-grbl

https://boxzy.dozuki.com/Answers/View/581/How+to+get-make+a+postprocessor+for+ArtCAM.

MeJasonT, 2019-05-06:
Sorry guys, I got lost in the ether.
My links not opening, yep i get that it happens all the time, have no idea why.  I follow the rules and paste the links into the message and they screw up.  It must be some sort bizarre magic.  when it happens to me on other threads i just copy and paste link into a browser and it works fine, to be honest its probably the best way to do it, Just because you see a link does not always mean that  it will take you there.  I could have linked you to Dimond Lils and renamed the link to anything i like.  Don't you just love computers.  I didn't by the way - just saying.

ntd, 2019-05-15:
@MeJasonT You must paste your link into the round brackets (_url_ is just a placeholder) and put between square brackets what you want to be visible, e.g. `[link to GitHub](https://github.com/)`. See [this article](https://help.github.com/en/articles/basic-writing-and-formatting-syntax#links) for further info.