gnea/grbl #656: X stepper only in one direction

schwittlick, 2019-05-04:
Hey guys, I have a trouble that my X stepper is only going in one direction.

I am using the GRBL-Plotter frontend to control it. When I'm using the Jog function I can move the stepper in both directions. Only when executing gcode it only moves in the positive direction.

My grbl config:

```
Grbl 1.1g ['$' for help]
[MSG:'$H'|'$X' to unlock]
$0=10
$1=255
$2=0
$3=4
$4=0
$5=0
$6=0
$10=0
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=500.000
$101=500.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=500.000
$130=200.000
$131=200.000
$132=200.000
```

Here's some simple rectangle gcode that has the problem:

```
G54 (Setup - GCode-Header)
M3 S20000 (Start spindle - Option Z-Axis)
G00 Z2.000 ( -1 ) 
G00 X-30.000 Y-30.000 ( -1 ) 
G01 Z-1.000 F400 ( 1 ) 
G01 X-30.000 Y-10.000 F500 ( 1 ) 
G01 X-10.000 Y-10.000 ( 1 ) 
G01 X-10.000 Y-30.000 ( 1 ) 
G01 X-30.000 Y-30.000 ( 1 ) 
G00 Z2.000 ( -1 ) 
G00 Z2.000 ( -1 ) 
M05 (Finish - Spindle off)
M30 (Setup - GCode-Footer)
M30 (Program end)
```

On the line G01 X-30.000 Y-30.000 ( 1 ) , it should go back to the originating position, though, on the machine moves into positive direction. The gui feedbacks the correct movements.

I would not be surprised when it would fail the same way in jog mode, though I can move all steppers in all directions when jogging.

Does anybody have a lead?

Thanks!

JamieAKing, 2019-05-04:
Assuming you are in absolute mode (G90), jogging usual happens in incremental (G91) some controllers do not switch back to G90 after jogging. Try prefixing your program with G90

schwittlick, 2019-05-04:
Thanks for the hint, but that's not it. As the Y axis is working fine with the gcode above. Even without prefixing with G90.

109JB, 2019-05-04:
What drivers are you using?  Many drivers reguire a delay between changing direction pin state and initiation of step pulses.  This can be set in config.h. Look for step pulse delay.

Also, strange as it may seem, try inverting the step port mask for X ($2=1). Doesn't seem like it should make any difference but I have had finicky drivers that needed it.

schwittlick, 2019-05-05:
I'm using ST-M5045 stepper drivers. 

I tried with different setting for STEP_PULSE_DELAY (10, 50, 100) and nothing changed. That's what you are talking about right?

And changing to $2=1 had no effect, even though I'm super open to these kind of magical reasons.

After checking again, I just really don't understand it. I've tried with another arduino uno, same situation. Jogging works in all directions. Gcode absolute positioning does not..

KhanhTrinh2094, 2019-05-05:
Same problem.
I'm using 2 A4988 driver, CNC shield v3, Arduino uno r3 smd
Y stepper working fine
But X stepper is only going in one direction

Occurs on both v0.9 and v1.1
I have no ideas. Tried to another cnc shield and another a4988
But still occurs

KhanhTrinh2094, 2019-05-05:
I bought the new arduino. and it works well
waste 1 week trying to fix the error
Just replace the Arduino and it will work

schwittlick, 2019-05-06:
Hey @KhanhTrinh2094, but that's not it. 

I have finally found out, what it is, it's the gcode.

After removing the gcode header and M commands, it's working fine:

```
G00 Z2.000 ( -1 ) 
G00 X-30.000 Y-30.000 ( -1 ) 
G01 Z-1.000 F400 ( 1 ) 
G01 X-30.000 Y-10.000 F500 ( 1 ) 
G01 X-10.000 Y-10.000 ( 1 ) 
G01 X-10.000 Y-30.000 ( 1 ) 
G01 X-30.000 Y-30.000 ( 1 ) 
G00 Z2.000 ( -1 ) 
```

closed 2019-05-06 by schwittlick

109JB, 2019-05-06:
On your G-code header...  I usually put in a header that covers all modal states to make sure they are set how they need for my program.  For example, my usual header is:

G17 G20 G40 G49 G54 G80 G90 G94

This insures that everything is set at the very start of the program.  Notice there is a G90 in that line so that absolute programming is set.  Even if it didn't fix the problem it is good practice.

I am a little unclear. In your opening post you said:

> On the line G01 X-30.000 Y-30.000 ( 1 ) , it should go back to the originating position, though, on the machine moves into positive direction. The gui feedbacks the correct movements.

Do you mean that when the machine moves the GUI feedback always matches the movement of the machine, or do you mean that when the machine moved the wrong direction that the gui feedback showed where it should have gone?

schwittlick, 2019-05-06:
Hey @109JB,

thanks for the tips with the good practice on the headers, I totally agree and check these out.

Yes, I mean that the GUI feedback did NOT match the movement of the machine. The movement in the GUI was as expected, the movement of the machine was wrong.

jasonw85, 2022-10-11:
I had a similar issue using a Arduino mega and a screw terminal breakout module.  X axis would in both direction while Y and Z only worked in +.  Issue was with the ground wires going to the stepper controllers.  They grounds on the arduino were not all connected, once I added an extra grounding wire the issue was fixe.