gnea/grbl #659: Homing and laser control don't work on bachin  maker 2 axis board

kryts, 2019-05-08:
I have a Bachin maker engraving board. it is running grbl 1.1z
with this version of grbl the jog commands do not work. it will accept the commands but nothing happens.
so i compiled and uploaded the latest version of grbl. This allows the jog commands to work. however now the laser control and auto home don't work.

109JB, 2019-05-08:
Since the current official version of grbl is 1.1g, the 1.1z version must be a custom version. Perhaps by the maker if your machine. Sounds like the complaint should be placed there.

kryts, 2019-05-08:
I did send them a email. I put it here more as if some one else has this board and knows of a fix. example maybe there is a change you need to make in config.h to make it work or one of the other files, then they could chime in. I totally agree with you that it is a custom version.

kryts, 2019-05-08:
[image]

kryts, 2019-05-10:
well I received a reply from the board manufacture and they are going to see if they can make a version for me that will work. I will let you know how it goes.

kryts, 2019-05-11:
They asked for pictures of the board to verify what one it was. The response I got back was that it is using the standard grbl pins. so i'm not sure where to go from here.

kryts, 2019-05-12:
ok got it compiled again and re-uploaded. it was an error on my side. i was editing the wrong config.h file

closed 2019-05-12 by kryts