gnea/grbl #661: Backlash compensation

shooter64738, 2019-05-09:
I have added backlash compensation to my grbl project if anyone is in need of it. Its compiled for the mega 2560 version. The amount of backlash compensation can be specified per axis. It does not compensate the first motion made after start up. Once a motion has executed any motion after that in which the axis direction changes, the compensation will be added. Steps used as compensation are not reported as part of the '?' position report.
[Grbl 2560 backlash project](https://github.com/shooter64738/grbl-mega)

Schildkroet, 2019-05-09:
As far as i've seen, you just add some steps to the planned move which will mess up the interpolation. The bigger the backlash, the bigger the interpolation error.

shooter64738, 2019-05-10:
I think I have addressed that, unless you see something new that I have missed.

bwanamukubwa, 2019-05-28:
Hi, would it be possible to have a short video of the features you have contributed? 

I am interested in the backlash compensation and have a controller i'm willing to put a 2560 on and any user guide is much appreciated. 

Thank you in advance.

beautifulsmall, 2019-07-19:
Hi, Im seeing movement on other axis. Moved all my settings from an UNO to a mega . There are new parameters 140,,141,142. I guess these are the backlash ? no description.  I've set all to 0.000 .
I have an XY setup only.  I tried to jog with Grbl-plotter and had cross over between the axis. It was ok for the first few jogs then the wrong axis starts moving before the correct axis moves the correct distance? the system was working with the UNO . It is so repeatable its not electrical crosstalk. 
IDE 1.8.4, TB6600 drivers. No setting changed in config.h . love this fork, the best upgrade yet . I will try the main grbl-mega over the weekend. let me know if you want the parameters i have set.

shooter64738, 2019-07-19:
Yes 140,141,142 are the distances for each axis backlash value. Right after I published that I ended up in the hospital and havent had much time to look at it. I had tested it and it seemed to be working for me, but its possible theres a problem. If you would open an issue on the repo branch for it, I'll take another stab at it and see if I can duplicate what you have and work out whats might be wrong.
thanks!

khturnings, 2020-06-29:
Will backlash compensation be included in the standard GRBL project to run on an UNO.
Has anyone already tried it on an UNO?
I am sure this will be useful for a number of novice builders who have built machines that would benefit from this like me.

sinfocomp, 2020-06-29:
No space for code

khturnings, 2020-06-29:
Oh - ok thanks @sinfocomp

yadavrp, 2021-08-23:
I wants to add the backlash correction to GRBL on mega2560 for 4 axis version 1.1f for my machine i am having two motors on X axis . it will be help full if you can guide me which files I have to modify (only the file names will be ok) to add this feature to my version of GRBL . I am expecting the following files 1. addition of files backlash.c and backlash.h  2. modefying the code in files planer.c and planer.h.  Please suggest if any other file is also needed to be checked and modified.

shooter64738, 2021-08-23:
If i recall its, Motion.c, Planner.c, Grbl.h
And include the backlash.c/h files.
I put most of my time into a different control since then and its been some time since i reviewed that change.

yadavrp, 2021-08-23:
thanks @shooter64738 I will try these files

nsiatras, 2023-01-08:
I' ve successfully implemented backlash compensation to an ESP32 Grbl flavor.
The code is here: https://github.com/SourceRabbit/Rabbit_GRBL

The backlash compensation algorithm takes place in
[Backlash.cpp](https://github.com/SourceRabbit/Rabbit_GRBL/blob/main/Grbl_Esp32/src/Backlash.cpp)
[Backlash.h](https://github.com/SourceRabbit/Rabbit_GRBL/blob/main/Grbl_Esp32/src/Backlash.h)
and 
[Stepper.cpp](https://github.com/SourceRabbit/Rabbit_GRBL/blob/main/Grbl_Esp32/src/Stepper.cpp)

vlastikcocek, 2023-02-17:
hi, this simple edit edit works for me.
I omit the calculation of the specified number of axis displacement cycles
[image]

ChubbyBunny1, 2023-09-28:
> hi, this simple edit edit works for me. I omit the calculation of the specified number of axis displacement cycles [image]

Trying to implement on the Z axis. I have the latest version of GRBL, where the variables should be declared in the stepper.c file? Did you define the variables outside of the ISR struct?