gnea/grbl #663: Sketch too big with gcc 9.1.0

ntd, 2019-05-13:
When compiling a customized GRBL 1.1f on ArchLinux current, I get a _sketch too big_ error (110% memory used). To be on the safe side, I tried to build a plain grbl-master image and this is the error message:

```
...
Sketch uses 35492 bytes (110%) of program storage space. Maximum is 32256 bytes.
Global variables use 1633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Arduino/Genuino Uno.
```
Downgrading `avr-gcc-9.1.0` to `avr-gcc-8.3.0` solves the problem. Building the same sketch gives:
```
...
Sketch uses 29786 bytes (92%) of program storage space. Maximum is 32256 bytes.
Global variables use 1633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
```

SL-RU, 2019-07-01:
Same problem here...

larsch, 2019-07-17:
I reported the issue on GCC bugzilla: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=91189

Raniita, 2019-07-24:
Same problem here. Fixed downgrading to avr-gcc-8.3.0

agajdosi, 2019-09-14:
Same problem here, any update on this?

zygmunt, 2019-09-15:
Is this really gcc bug? See here my bug report on lufa: https://github.com/abcminiuser/lufa/issues/149
BTW I'm taking the same problem with gcc9

cramsted, 2019-09-28:
Found an easy workaround. The problem doesn't exist with the Arduino IDE 1.8.10, however, as of this writing, it isn't currently available on pacman. The easiest way around it is to uninstall all of the Arduino software on your system, go to the [Arduino downloads page for linux](https://www.arduino.cc/en/Main/Software), download 1.8.10, unzip it and then run the install.sh script.

wmarkow, 2019-12-06:
@cramsted, your workaround helps because Arduino and its IDE supports **avr-gcc** at most in the version of **7**. However I understand that somebody want's to compile it with a newest version of **gcc**.

Here https://github.com/qmk/qmk_firmware/issues/6719 they have the same problem.

wmarkow, 2019-12-06:
I was able to reproduce this issue as @ntd desrcibed. I was using [Sloeber with Eclipse](https://eclipse.baeyens.it/) to compile (but midified a bit to use avr-gcc-8.3.0 and avr-gcc-9.1.0). Indeed the text size is increased from **29790** to **35496** bytes.

Please take a look at the attached [grbl_disassembly.zip](https://github.com/gnea/grbl/files/3932487/grbl_disassembly.zip) file It contains:
 * grbl_gcc_8.3.0_build.log from the build with gcc 8.3.0
 * grbl_gcc_9.1.0_build.log from the build with gcc 9.1.0
 * grbl_gcc-8.3.0_text_section_info.txt with just a text section info (how big is every function) 
 * grbl_gcc-9.1.0_text_section_info.txt with just a text section info (how big is every function). You can compare it with the grbl_gcc-8.3.0_text_section_info.txt using for example [Kdiff3](http://kdiff3.sourceforge.net/) software. You will see that some function are bigger with *gcc-9.1.0** than with **gcc-8.3.0**
 * grbl_gcc_8.3.0_disassembly.txt with a full disassembly of the code. It contains assembler + C code mixing.
 * grbl_gcc-9.1.0-disassembly.txt with a full disassembly of the code. It contains assembler + C code mixing.

You can easily compare **grbl_gcc_8.3.0_disassembly.txt** and **grbl_gcc-9.1.0-disassembly.txt** to see how it was compiled. 
For example the **printFloat** looks interesting.
With **gcc-8.3.0** we have (a casting to **long** and to **uint32_t**?):
```
uint32_t a = (long)n;
     7ce:	0e 94 de 36 	call	0x6dbc	; 0x6dbc <__fixsfsi>
```
but with **gcc-9.1.0** we have:
```
uint32_t a = (long)n;
     932:	6e 87       	std	Y+14, r22	; 0x0e
     934:	7f 87       	std	Y+15, r23	; 0x0f
     936:	88 8b       	std	Y+16, r24	; 0x10
     938:	99 8b       	std	Y+17, r25	; 0x11
     93a:	6e 85       	ldd	r22, Y+14	; 0x0e
     93c:	7f 85       	ldd	r23, Y+15	; 0x0f
     93e:	88 89       	ldd	r24, Y+16	; 0x10
     940:	99 89       	ldd	r25, Y+17	; 0x11
     942:	0e 94 eb 41 	call	0x83d6	; 0x83d6 <__fixsfsi>
     946:	6e 87       	std	Y+14, r22	; 0x0e
     948:	7f 87       	std	Y+15, r23	; 0x0f
     94a:	88 8b       	std	Y+16, r24	; 0x10
     94c:	99 8b       	std	Y+17, r25	; 0x11
```
It looks much complex when compiled with **gcc-9.1.0**.
In general I have noticed that **gcc-9.1.0** uses a lot of operations as above (it uses **ldd** and **std** on some registers) especially around and together with **call** operation related to math operations (_mulsf3, __floatsisf, __gesf2 ?).

Somebodey from **GCC** project could know more about it. Maybe it is related to the **call convention**? Maybe there is some switch/configuration parameter that need to be passed to the **gcc-9.1.0** compiler but was not required in **gcc-8.3.0**?

wmarkow, 2019-12-07:
The four **std** and **ldd** instructions in my previous comment, before **call** instructions seem to "do nothing". According to the [AVR Instruction Set Manual](http://ww1.microchip.com/downloads/en/devicedoc/atmel-0856-avr-instruction-set-manual.pdf), first the **r22 - r25** registers are stored somewhere in memory and then are restored back. No other instructions are in between, so it looks like those eight operations are not necessary; they make the final image to be bigger.

ntd, 2019-12-08:
@wmarkow You should try to submit your remarks to the [gcc bug](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=91189): I don't think the GRBL folks can do something about it.

wmarkow, 2019-12-08:
@ntd I will do that. Currently I'm waiting for my gcc bugzilla account (it needs to be created manually by administrator).

wmarkow, 2019-12-13:
I just found that a similar issue related to generating [overhead operations](https://github.com/gnea/grbl/issues/663#issuecomment-562588954) is already reported as https://gcc.gnu.org/bugzilla/show_bug.cgi?id=90706

Harvie, 2020-02-23:
I made this hex on old ubuntu. Hopefully it will help someone.
[grbl-v1.1h.20190825.hex.txt](https://github.com/gnea/grbl/files/4242228/grbl-v1.1h.20190825.hex.txt)

goosenphil, 2020-04-17:
I've experienced the same issue using GCC AVR 9.2.0

I've been able to compile it on Fedora 31 by using the ```toolbox``` command to create a Fedora 30 container. Here are the commands if you want to do the same (performed in your grbl directory):

```bash
toolbox create --release f30
toolbox enter --release f30
sudo dnf install avr-gcc-7.4.0-5.fc30 avr-libc make
make -j
```

The result of compiling with GCC 7.4.0 in the container:
```
avr-size --format=berkeley build/main.elf
   text	   data	    bss	    dec	    hex	filename
  29586	      0	   1633	  31219	   79f3	build/main.el
```

dzervas, 2021-10-20:
Another workaround I found was using ubuntu 20.04 which uses gcc 5.4.0:

```bash
docker run --user ${UID}:${GID} --rm -it -v ${PWD}:/mount rubberduck/avr make grbl.hex
```

ArthurFleischman, 2022-06-28:
avr-gcc error compiling
```
avr-gcc -Os -DF_CPU=16000000UL -mmcu=atmega328p -l /usr/lib/libisl.so.21 main.c -o main.out
/usr/share/avr/lib/gcc/avr/8.3.0/cc1: error while loading shared libraries: libisl.so.21: cannot open shared object file: No such file or directory
make: *** [Makefile:9: main.out] Error 1
```
id doesnt find the libisl.so

theminjus, 2023-06-09:
Had the same problem with compiling on the last version, I had the compiler set on AVR ISp and after compiling the code it said 161% over memory. 
Tried with AVRISP mkll and got compiled under 95%.

Change it to AVRISP mkll on Programmer and it will work.

nokados, 2023-07-22:
Docker workaround works great. There is a special [docker image](https://hub.docker.com/r/tiryoh/grbl-builder) for building grbl. Just go to grbl root directory and run
```
docker run --rm -it -v "${PWD}":/source tiryoh/grbl-builder:latest make
```
After that, grbl.hex will appear in this directory. So, you only need to [flash](https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino) it to your board. For example, for Arduino Nano (old bootloader):
```
avrdude -v -D -c arduino -P /dev/ttyUSB0 -p atmega328p -U flash:w:grbl.hex:i -b 57600
```