gnea/grbl #664: Ideal pin to use for plasma control relay GRBL 1.1 and greater

Slixxor, 2019-05-20:
We are making a versatile hardware controller that will make it easy to connect spindle, laser or plasma. we've got a transistor output for spindle and laser control, that's fine.

What we are trying to do is feed relay contacts to switch the plasma on and off. with our custom board the relay will be bound to it's respective pin permanently without user interaction. 

The question is, should we bind the relay to Spindle enable, Spindle direction or Spindle PWM.

Obviously we don't want the relay to chatter during normal spindle or laser operation. We were thinking of adding an ATTiny to receive the pwm signal on spindle PWM pin (If any) and then create a solid 5V output. Although we were hoping to avoid this.

We will have independent screw terminals for Spindle, PWM laser and Plasma on the custom board.

So, What do you think the best option would be to eliminate the need to change any hardware and be able to control spindle, laser or plasma.

MeJasonT, 2019-05-20:
Im quite happily running a plasma table using the spindle pwm pin to drive the run/burn relay. I set the spindle speed to 1000 which provides the max pwm output.  I also use the suds/coolant pin to trigger a pilot arc relay.  The post processor in cambam has been written to delay for 1 second when the torch reaches its start location and moves down to Z0 (stock surface).  The pilot arc is triggered then there is a delay of 0.5 seconds then the run/spindle command is enabled a delay of 1.5 seconds is then given to allow the arc to punch through the material after which the machine is allowed to move off and the pilot relay disengaged.   So yes you can use the pwm pin.  Bear in mind that the high frequency noise causes no end of issues when the arc is struck - good shielding and grounding of all cables and machine must be carried out - if you do get problems.  A particular one is false triggering of the end stops.

Slixxor, 2019-05-20:
> Im quite happily running a plasma table using the spindle pwm pin to drive the run/burn relay. I set the spindle speed to 1000 which provides the max pwm output. I also use the suds/coolant pin to [...]

Hi Thanks. The concern here is when using spindle mode or laser mode. the relay will be bound to spindle PWM on our board. The relay will chatter if S is less than 200. 

We are using a low frequency pilot arc plasma cutter at present so EMI and ground noise is minimal.

As far as noise goes, we've already sorted that on the new board. We use a TLP281 4ch opto-isolator IC for the limit switches, have a mains filter on the PSU Input and Ferrite sleeves on limit switch and stepper cables. 

We discovered from our testing that the main issue with the high frequency machines we've tested was in fact DC ground noise on the controller input. 

We've created a resistor, cap & diode network on the DC input side of the controller. From analyzing data on the cro, we've gone from 3.5V of fluctuation on the DC ground to 0.03mV by doing this. Most people focus on a smooth DC positive input and ignore smoothing out the DC ground.

Also, depending on the power supply used, i.e: a Meanwell PSU, you'll find the DC ground and mains ground are not shorted. By shorting the DC ground to the mains ground you can stop the DC side "floating". This has helped a few of our other projects greatly. BE VERY CAREFUL TRYING THIS! Some PSU's may literally explode if you short the DC ground to mains ground.

Not sure why some PSU's like to keep the DC ground floating.

MeJasonT, 2019-08-26:
Have you considered a jumper link on your board which would bypass the relay and direct the output to  the same wiring terminal for laser/spindle o/p.  Overkill having a relay installed in the board if its not required for some users but hey it will provide the option you are looking for.  in any case the RPM min/max of the system will have to be changed between option selection as all three (laser,plasma,spindle) will have different operating parameters anyways.  For example it would not be easy to have a 3 axis machine where you can hot swap tooling (laser,plasma,spindle) and have the machine quite happily run the job, for starters your cam application will set up jobs differently for different tasks.  Milling would have some element of depth related to it where the laser and plasma are set at a pre determined Z height and run the job from there. In my case the plasma has a pre determined startup block in the post processor which triggers the pilot ark, waits then triggers the cut (disabling the pilot after a pre determined delay time).  The pilot is triggered long enough for the plasma arc to be generated and penetrate the work piece then disabled to prevent it eating consumables which will happen if not deactivated - its effect is like constantly dragging the torch on the work piece.

Slixxor, 2019-09-26:
> Have you considered a jumper link on your board which would bypass the relay and direct the output to the same wiring terminal for laser/spindle o/p. Overkill having a relay installed in the board [...]

We've opted for Spindle_Enable line. It's very consistent and means no potential chattering.

closed 2019-09-26 by Slixxor