gnea/grbl #666: Only reply "ok\r\n" only after Grbl has completed the current line of Gcode?

oliver-ernst, 2019-05-23:
Hello,

I'm just doing a university project and am using a number of MCUs with GRBL all running together and am trying to keep them in sync.  I'm wondering whether there is a setting anywhere so that GRBL finishes the current line of Gcode it has been sent before replying with "ok\r\n".  I've noticed that once the motion planning buffer is full this happens, but until that point it replies ok as soon as the line is read in and added to the buffer.  

I did see and try to change without luck the BLOCK_BUFFER_SIZE variable in the planner.h file, 'm just wondering if anyone has encountered this problem or if there's any quick fix for it?  Or if I'll have to change the code substantially to make it work?

Thanks for your help.

109JB, 2019-05-23:
If the syncing only has to occur at certain times then you can insert a G4 P0 line.  For a G4 command the buffer will not accept any additional commands and an 'ok' response will not be generated for it until the G4 completes.

Also, the status report has a section reporting buffer state that looks something like Bf:15,128
This has to be enabled in config.h and in the $10 status report mask.
With this you could write a sending program that checks that the first number = 15 before sending the next command.

oliver-ernst, 2019-05-23:
Thank you so much, that's great information!

closed 2019-05-23 by oliver-ernst