gnea/grbl #668: Stepper driver DVR8825

peagerfw, 2019-05-23:
Hi,i have new stepper driver dvr8825 but if i engrave 40x40mm size final product is cca 20x20mm
If use stepper driver A4988 engraved size is ok.Please how to set DVR8825?
Thank you.

109JB, 2019-05-23:
You need to adjust your steps/mm. See this:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm

peagerfw, 2019-05-23:
i adjusted $100, $101 and $102 – [X,Y,Z] steps/mm from 800 to 1600 but stepper motor stop after 20mm and buzzer it

109JB, 2019-05-23:
post all of your settings

peagerfw, 2019-05-25:
$0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=0.000
$30=10000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=1800.000
$111=1800.000
$112=1800.000
$120=100.000
$121=100.000
$122=100.000
$130=300.000
$131=150.000
$132=44.000

JamieAKing, 2019-05-26:
You need to reduce steps per mm not increase, alternatively alter the dip switches on the drive itself to match the current setting. Looks like you need to increase micro steps from 4x to 8x 
The stall and stepper buzz could be lack of current from the new driver? Have you adjusted this to your motors rating?

MARIOBASZ, 2019-05-26:
you must control the micropasses that you have established with the m1 m2 m3 bridges in drv8825, you are probably giving more micropasses than in A4988, look at the technical data of each one

peagerfw, 2019-05-27:
Thank you so much for help
https://github.com/arkypita/LaserGRBL/issues/664