gnea/grbl #669: Stepper driver not holding when DEFAULT_STEPPER_IDLE_LOCK_TIME = 255

oliver-ernst, 2019-05-25:
Hey guys,

You've all been great to me so far, just having another little issue with GRBL.  I'm running it on the Atmega2560 and am trying to hold my motors when not sending commands.  I've set the DEFAULT_STEPPER_IDLE_LOCK_TIME to 255 because I've read this should lock them in place when not being used, however it's not currently working.  I've checked the pins with a logic analyser and am only getting a PWM signal when motors are moving.  

There's no PWM signal to hold the motor in place which it should, shouldn't it?  Just wondering if anyone else has had success getting motors to hold and if there's anything else I need to change in the GRBL settings/code to get it to work.

Cheers.

109JB, 2019-05-25:
Steppers will be on anytime the enable pin is at the appropriate level, usually low. It will not have pwm though, only high or low.

oliver-ernst, 2019-05-25:
Thank you so much, I might try and just set the pin manually then.  Cheers.

closed 2019-05-25 by oliver-ernst

109JB, 2019-05-25:
Are you using steppers? If yes, there is no reason to manually set anything or change the code itself as everything works just fine for steppers.

The enable pin on most stepper drivers is low for enable, high for disable, so simply leaving it unhooked will also leave the steppers enabled all the time.  You can set DEFAULT_STEPPER_IDLE_LOCK_TIME in the EEPROM configuration settings. It is $1.