gnea/grbl #670: I need a software for cnc wood lathe

samehherit, 2019-05-27:
Hi experts ;
 I need a software for cnc wood lathe  3 or 4 axis, 

Any suggestions ? 
Thanx for helping .

JamieAKing, 2019-05-27:
For simple profile turning I see no reason why grbl would not be suitable, difficultly begins with threading/ turning spiral spindles etc where z movement needs to be synced with the spindle rpm. 
Clough42 on YouTube is currently tackling this to give his manual lathe an electronic lead-screw for threading, https://youtu.be/FTs9GygRQ-U
I’m sure something like this could be incorporated in grbls firmware or the controller gui but it’s way above my pay grade

109JB, 2019-05-27:
Spindle sync for threading already has been added for the Grbl Mega version

https://github.com/fschill/grbl-Mega/tree/spindle_sync

https://www.youtube.com/watch?v=Ia_g6kpPqtE&feature=youtu.be

samehherit, 2019-05-31:
Thanx alot for helping

MeJasonT, 2019-05-31:
I am currently pondering the conversion of my lathe to cnc.  Still trying to get my head around how tool paths are created for both lathe and "A" 4th axis before i commit to overhauling my machine.  Looked at a few examples on cambam but its still white mans magic, there appears to be very few people using GRBL on lathes so finding worked examples on google is difficult to say the least.

109JB, 2019-05-31:
Not sure I understand what you are saying. A 4th axis and a lathe are 2 different things with a lathe being only 2 axes, X and Z. The axis orientations are different, so a 4th axis program won't work on a lathe and vice-versa.  

I have used Grbl for a pseudo-lathe, that being my 3-axis milling machine with a lathe chuck mounted to the spindle and the lathe tool held in the vise on the table. This configuration results in orientation of the X and Z axes just like they would be on a lathe. Moving Z+ moves the tool away from the chuck, and moving the X+ moves the tool farther from the spindle centerline. The Y axis is only used to center the tool. The only thing different with this configuration from a lathe is that as a lathe in this manner the X is in "radius" mode so the moves will be based on radius rather than diameter. If your lathe g-code program uses radius mode it will run just fine in a mill used this way. You could change this by simply setting the steps/mm to double the current value when using as a lathe.  Here is a link to a video showing this:

https://www.youtube.com/watch?v=tG3NLPtEDsw

JamieAKing, 2019-05-31:
@109JB  10 points to grifindor, nicely done sir

MeJasonT, 2019-05-31:
Hi John, 
I was making the asumption that most questions on the GRBL site relate to milling machines where an additional rotary axis would become A axis (the fourth axis - depending on orientation of machine of course. along x axis = A  along y axis =B etc.) is the chuck on a cnc lathe not classed as the A axis ?
Stop being a smarty pants and finish your GUI we are all waiting to try it out.
Like the video, really impressive.

109JB, 2019-05-31:
On a CNC lathe, Z axis moves parallel to the spindle axis and X axis moves perpendicular 

[image]

109JB, 2019-05-31:
Wanted to point out that the video isn't mine. Just a video I picked showing what I have done.

terjeio, 2019-06-01:
[Ezilathe](https://www.cnczone.com/forums/uncategorised-cam-discussion/263938-cnc-cam-forum.html) might be a useful tool for generating toolpaths. Registration is required for [download](https://www.cnczone.com/forums/downloads.php?do=file&id=21).

closed 2019-08-02 by samehherit