gnea/grbl #673: grbl use with MKS BASE 1.6 board

scmaher, 2019-06-04:
I am using grbl on the MKS BASE 1.6 board.  It is basically an arduino mega combined with a RAMPS 1.4 on a single board.  Currently available for around $30 (with 5 A4982 drivers installed), so its pretty attractive.

The first problem I encountered was that the microstepping settings were different for the various axis.  On the RAMPS 1.4 board these settings are controlled via jumpers.  On the MKS BASE board, the MS1 and MS2 pins on the A4982 drivers are routed to pins on the atmega mpu:
             
X     MS1     PE2 (4)
       MS2     PH3 (15)
Y     MS1     PF5 (92)
       MS2     PF4 (93)
Z     MS1     PA0 (78)
       MS2     PG2 (70)
E0   MS1     PK1 (88)
       MS2     PK2 (87)
E1   MS1     PF3 (94)
       MS2     PG5 (1)  

As far as I can tell, these pins are not initialized anywhere in the grbl code.  Its not reasonable to expect grbl to account for every variation of board in the world.  I've added some code to steppers.c to initialize these pins as outputs and set them to the correct state during the rest of the stepper initialization.  

However, this doesn't seem like a very sustainable solution, as I'd like to be able to easily take updates as they become available.  I know that I can add custom board definitions to the various .h files, but where should I locate board-specific code?  

The second issue isn't really an issue, just a note for others who might be struggling with the problem.  The MKS BASE 1.6 board has limit switch inputs for both min and max limits for x, y, and z axis.  However, grbl reports only a single pin state variable for each axis.  If you define the limit switches to be active high (a positive voltage on the limit pin indicates that that limit switch has been activated), any unconnected limit switch inputs can mask the connected input.

As an example, I have a X- (minimum) limit switch, but no X+ limit switch.  Since grbl reports only one pin state for X axis limit switches, it always reports "active" for the X axis, regardless of the state of the X- limit switch.  

The solution to this is simple, I just added a jumper between the limit pin and ground (conveniently located next to each other on the board I am using) for each limit switch input I am not using.

GRBL is an outstanding project, I've been using it since the early releases!

Steve

109JB, 2019-06-05:
> I am using grbl on the MKS BASE 1.6 board. It is basically an arduino mega combined with a RAMPS 1.4 on a single board. 

Are you using the Mega version of Grbl? Because you are in the Grbl for 328p issues forum.

https://github.com/gnea/grbl-Mega/wiki

scmaher, 2019-06-05:
Sorry, my bad...

closed 2019-06-05 by scmaher

Ferz82, 2024-10-06:
Hi. I have mks base 1.4.
Please help me set up grub-mega or grbl-mega-5x.
Can I see your files and settings?
Or maybe post your ready-made firmware?
Thanks in advance!