gnea/grbl #675: Woodpecker 3.2a with 3 pin laser and laser GRBL

spacetrucking, 2019-06-09:
Hi,
I've been using a woodpecker GRBL 0.9 to control my CNC miller and recently purchased a chines 3 pin laser, when I recieved it I released that the woodpecker 0.9 would not support the 3 pin laser so I bought the woodpecker GRBL 1.1 which does support the laser.
I'm also using for the first time Laser GRBL, 
I have two questions, the first I think I have answered myself.
the woodpeck laser pinout is 12v - PWM - GND and the laser pinout is 12v - GND - PWN if I connect the laser up as you would think 12v - PWM - GND the laser is on all the time even whilst running Laser GRBL, if I swop the PWM and GND the laser holds off but the fan speed modulates as if it was varying the laser power. If i put a multimeter on the PWM and GND pins the voltage varies between 0 and 5V as you would expect. I just not sure I believe the chines laser pin out..

second question, if I have connected it correctly and laser is on all the time how do I get it to control the laser ? I have set $32 = 1 to enable the laser but do I need to do anything with $30 and $31 ??

any ideas
Regards

Paul

spacetrucking, 2019-06-11:
So I've been doing a bit more investigation and when I create the G code with LaserGRBL the first lines of code set M3 S255.
which I gather sets spindle speed to 255 or full speed, I guess this effectively turns the laser on to full, but I can'nt work out why the laser turns on straight away ??
I have set $30 = 255 i.e max speed
and $31 = 0 i.e minimum speed

spacetrucking, 2019-06-11:
I have had a small break through
if I connect my woodpecker board PMW output to my laser GND and set
$30 = 0
$32 = 255
the laser works irrespective of $32 set to 1 or 0
although I don'nt know if the power is modulating which I don'nt think it will be 

has anyone any thoughts ??

gareth86, 2019-07-05:
Hi, did you get this sorted as im having the same issue???

spacetrucking, 2019-07-06:
Hi
I sort of solved it, I sent the laser back to eb....y and bought a second
one which worked. I bought one of these,

 *Focusable Analog & TTL 5.5W 5500mW 450nm Blue Laser Module Engrave Cutter
+ Free Goggles.*