gnea/grbl #676: i need your suggestions

samehherit, 2019-06-10:
Hi every body 

 i need your suggestions about an electronic or mechanical tool which i can install it on the (x ) axis to scans the 3d models as an input data to transfer it to artcam or any apps which i could engrave this model later
It just a dream ind i hope to become true

MeJasonT, 2019-06-10:
https://www.david-3d.com/en/support/downloads
use meshcam or blender to create your dxf for production of job file.
Its based on a camera image scan, if the camera was mounted so it moved across the object like a paper scanner this would produce a 3d image of the object beneath all be it not the side sitting on the machine bed.   There is a demo version that works with a webcam.

samehherit, 2019-06-11:
That is so good 
Thanx a lot for your helping 😊

closed 2019-08-02 by samehherit