gnea/grbl #679: Woodpeaker V3.3 Limit Switches

raindancer2204, 2019-06-25:
Hello I have a Woodpeaker CNC motherboard version **V3.3**.

I had a Versio **0.9 before** and here I had 2 connections for the end stops for the axles.

Can someone please tell me where I have to connect my 6 individual endstop switch connections now?

I have pins down right for X and Y and Z. But where do I put the other wires? I have 2 for the axle?

Thank you Raini

brawd, 2019-08-27:
I'm not absolutely familiar with the MoBo V3.3 but if it's a newer job you have probably lost the 6 pins for limit switches and now have 3 pins, plus of course the earth pins that go with them.
If this is so then the X switches, the Y switches and the Z switches are connected in parallel. There is further information available on a web search.
hope it helps,
brawd

ulab, 2019-10-22:
It depends if you had used NC (recommended) or NO switch setups:
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches