gnea/grbl #681: QUESTION. $ commands with offline controller

fedekrum, 2019-06-27:
Hi.
I tried to home my CNC ( $H ) at the beginning of a file that I put to work in an offline controller.
Did not work !!

The offline controller is a typical one. 

Any idea if $ commands can be executed from a file?

109JB, 2019-06-27:
That cannot be answered because it is totally dependent on how the streaming code is written and if it allows $ commands to be sent from within the file.

chamnit, 2019-06-27:
@109JB is correct. That is a function of your streaming code. Not Grbl. I also would add that it’s not a good idea to include homing cycle in your g-code program. It can easily result in crashes if you aren’t paying attention.

closed 2019-06-27 by chamnit

fedekrum, 2019-06-28:
@chamnit , you are right.
But I was planing to have a file with the only purpose to make a home, and run it before sending other files.

109JB, 2019-06-29:
Why ask?  Just try it and then you will know if it works or not.

fedekrum, 2019-06-30:
> Why ask? Just try it and then you will know if it works or not.

I tried (as you can read in my first post. But I was not sure if there where oder settings or something specific I had to tweak.