gnea/grbl #682: Can't disable Z axis homing

griegory, 2019-07-01:
Hi,
I have read several issues regarding this topic but I'm still not able to stop de Z axis from homing.
I have disabled it in the GRBL config.h file in the arduino library folder in this way:

// #define HOMING_CYCLE_0 (1<<Z_AXIS) 
// #define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))
// #define HOMING_CYCLE_2
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))

Once I save the config.h file I just open Arduino IDE (the grbl example shows up immediately) and I press upload. I also make sure the arduino is connected to the current and to my laptop through USB.

Then, when I open UGS and press the homing button the Z axis starts homing again.

What am I missing? my guess is I'm missing something when uploading the GRBL library again to the arduino. Is there a way to clear the arduino memory and install a library again from scratch?

Thank you all

closed 2019-07-01 by griegory

lucky62, 2019-07-02:
What was the problem? Explanation can be useful for others...

griegory, 2019-07-02:
Well, when I first downloaded the GRBL zip file I copied the GBRL folder into the Arduino library folder in my main drive/Program files, etc. which I thought Arduino IDE was using to upload the library to the Arduino.
I didn't realize that Arduino IDE was actually using another library that was created in documents/Arduino in my main drive.

I don't remember if I selected this folder or if Aruino IDE creates it. It would be good to know.