gnea/grbl #684: Weird behavior on Homing

BernardG, 2019-07-08:
I am switching from DRV8825 drivers to TB6600. Previously, with the DRV8825, I had homing doing exactly what it is supposed to do. Now, it's like it was "mimicking" homing, and that's it.  What I mean is that, if all 3 axis are close (+-10 mm) to the Home position, it looks good, hit the switches, and back up 2 mm, as setup.

BUT, if all axis are, say, 50 mm away from Home, it just do the same. Move a few millimeters, then act as if had hit the switches, move slowly, and back up a couple of millimeters, WITHOUT TOUCHING ANY SWITCHES!

I have been checking with Status, and the switches are working as expected. What could be the reason? Could electrical noise/interference be the cause? 

I am lost, thanks for pointing me in the right direction.

MikeNewyen, 2019-07-08:
I also got a similar problem; I set the $23=3 because I wanted to change the coordination to the bottom left. I also used the
#define HOMING_CYCLE_0 (1<<X_AXIS)  // COREXY COMPATIBLE: First home X
#define HOMING_CYCLE_1 (1<<Y_AXIS)  // COREXY COMPATIBLE: Then home Y
As I don't have the Z-axis, and have #define HOMING_FORCE_SET_ORIGIN as disabled. Now when I tried to home, it move in the opposite direction of the limit switch, and give me error 8 saying that it couldn't disarm the limit switch. Any guide would be appreciated.

chamnit, 2019-08-11:
@BernardG : You have electrical noise issues caused by your new drivers.

@MikeNewyen : Make sure your CoreXY system is setup exactly as the theory diagrams show, especially rotation direction with a programmed motion. If not, it can move all wonky.

closed 2019-08-11 by chamnit

BernardG, 2019-08-11:
Sorry I did not follow up earlier. Yes, it was electrical noise, and it's now solved. Thanks!