gnea/grbl #687: overload detection

alejandro-anv, 2019-07-16:
When working as CNC, is there any way to add an external sensor that informs grbl that feed rate is too fast and motor is overloading?
If not, can you point me to the part of the code that I can modify for adding this behavior?

The idea is that if the material is hard, the motion should be slower. A sensor can detect this and the software can modulate the speed acording to this mesurement (may be analog or digital on/off overload detected).

terjeio, 2019-07-17:
Use Trinamic drivers and [stallGuard2](https://www.trinamic.com/fileadmin/assets/Support/Appnotes/AN002-stallGuard2.pdf) load measurements as input to feed override?

alejandro-anv, 2019-07-17:
> Use Trinamic drivers and stallGuard2 load measurements as input to feed override?

The hardware is not the problem. I need a way to change the speed in grbl. 

My question is if there is already a way for altering the speed inside grbl or where in the code can I insert such change (i.e. multiply by a factor).

langwadt, 2019-07-17:
feed override, you can see how it is done in serial.c line 167-171

terjeio, 2019-07-17:
I wrote "input to feed override" -> plan_update_velocity_profile_parameters() in planner.c

closed 2019-08-11 by chamnit