gnea/grbl #688: Homing Issues

balinmitchell, 2019-07-18:
I have built a basic cnc running nema 23s and tb 6560 steppers controlled with grbl 0.9.
The limit switches are NC on a 24v circuit running to relays that connect to the 5v circuit of the arduino (also NC).
Having issues where the machine randomly errors (Error: Homing Fail) as it hits the switches, everytime it errors on a different axis and we have no idea what is going on. It used to work with seperate 2 core speaker wire for the switches running to the relays but I originally bought 6 core so I have joined the switches so that both the stepper motor wires and the switch wires are running parallel, yes I know this is a recipe for disaster in terms of interference but thats what it had before when the speaker wire was taped to the outside of the 6 core so nothing has changed, also it is running on 24v so shouldnt have interference issues?
The thing we have just tried to do is build an RC filter on the 5v side which hasn't done anything at all. With or without the filter there is a definite drop between 5v and 0v on the limit pins so not sure as to why it keeps erroring out?

emgalos, 2019-07-21:
Algo parecido me sucedio.  Siempre se disparaban las alarmas de posicion.  La solucion fue utilizar cable blindado para la conexion de los sensores de final de carrera y aterrizarlos todos a tierra y conectar el negativo de la fuente a tierra.

emgalos, 2019-07-21:
He tenido inconvenientes cuando cancelo una operacion de maquinado.  Al dar click en retornar a cero,  la maquina no sube primero el eje z sino que rettorna a  cero con el eje z en la posicion en que haya quedado con las consecuencias de dañar la pieza que se esta trabajando,  y una vez que llegan los ejes x e y a cero, entonces ahi si sube el eje z,  pero ya el daño esta hecho..  Como ya se lo que hace evito hacerlo de ese modo reseteando en software y buscando manualmente una posicion que me permita el retorno a cero sin problemas pero la mayor de las veces eso no es posible y pierdo las coordenadas.
Tengo una cnc hecha en elgarage y utilizo universal gcode sender 0.9.  Tambien aclaro que no soy experto en el tema,  mas bien muy curioso y con ganas de experimentar (Que bien costoso que sale jjejeje).  Utilizo arduino uno con grbl gnea 1.1f

JamieAKing, 2019-08-03:
@balinmitchell did you get this sorted? I realise this response maybe too late but as you have no useful replies I thought I may through some ideas at you. 
I had a lot of problems with emf from both my steppers and spindle. This resulted in the symptoms you describe. I have gone to great lengths adding shielding, filters and ferites and separating the signal lines. While I’m happy that homing works every time I still get an occasional false  hard limit whilst the spindle is running. 5 years into running the machine I’m confident enough to turn off the hard limits and live with it!

I’m no expert in electrical engineering and can only speak from my limited experiences.
 To get homing working working reliably I increased the voltage of the switches as you have but only to 12v and then ran that circuit into opto isolators rather than relays. I suspect the collapsing field in the relay as the limit switch opens coupled with the noise from the motor is the cause of your problem. A fly back diode across the 24v side of your relays may help. I wish I could help with part numbers but I bought a pack of 100 some 5 years ago and probably still have 90 left.
Something to try which may prove the above is to wire the limit switch’s normally open, so that when the gantry makes contact the relay energises. This is less noisy than the current set up but generally frowned upon as a fault in the wiring (broken connection etc) will not stop the machine  
With 24v it will take some fairly serious emf to energise a relay coil

balinmitchell, 2019-08-08:
> @balinmitchell did you get this sorted? I realise this response maybe too late but as you have no useful replies I thought I may through some ideas at you.
> I had a lot of problems with emf from both [...]

Thanks man, have tried all of this and some of it helped slightly but I think there are issues not worth fixing and the system we use instead is to manualy pull it against the limits before startup, this works well enough for what we need it to do.

Cheers for the help

closed 2019-08-11 by chamnit