gnea/grbl #690: Annoying work stop while processing

gquinet, 2019-07-20:
I have the camtool cnc board in my cnc 3018 pro and experiment very annoying errors in many jobs.
I use grbl control candle to send grbl code to the board and do jobs.
And very often, without any reason, the bord just stop all motors and brbl control don't do anything more, without any error in the console.

So, my cnc is just useless as i'm not able to finish any job i want to do. it's so frustrating, i'm thinking to put the board to trash and replace it by a simple arduino cnc shield.

It also looks like there no simple way to have access to the serial port to check/update binary.

I need help to solve this issue cause, i'm really about to drill holes in this board full of bugs and put it to trash can!!!

gquinet, 2019-07-20:
As a complément, it always happen on a diagonal movement with tool up.
More often on long moves just after moving tool up.

SailWithChips, 2019-07-20:
Hi gquinet, I would make one test. Lets your gcode run without motors connected and see if gcode file succesfuly ends. In this case then I would have deep look into emc or ‘power’ side in your system. Good luck!

gquinet, 2019-07-20:
hi! i tried without any motor connected and i got this error:
"ALARM: 23; [MSG:Pgm End]; ok" on the instruction "M3 S1000", but 5 instructions after that executed.
With another gcode file, no error but still interrupt after first circle.

Here is the gcode of the first file that stop after just few instructions:

`( Made using CamBam - http://www.cambam.co.uk )
( keys 7/20/2019 2:02:46 PM )
G21 G90 G91.1 G64 G40
G0 Z3.0
( top_keys2 )
G17
S1
G0 X109.778 Y93.3594
G0 Z1.0
G1 F150.0 Z-0.4
M3 S1000
G1 F110.0 X107.358 Y92.0657
G1 X104.731 Y91.269
G1 X102.0 Y91.0
G1 X99.2687 Y91.269
G1 X96.6424 Y92.0657
G1 X94.222 Y93.3594
G1 X92.1005 Y95.1005
G1 X90.3594 Y97.222
G1 X89.0657 Y99.6424
G1 X88.269 Y102.269
G1 X88.0 Y105.0
G1 X88.269 Y107.731
G1 X89.0657 Y110.358
G1 X90.3594 Y112.778
G1 X92.1005 Y114.899
G1 X94.222 Y116.641
G1 X96.6424 Y117.934
G1 X99.2687 Y118.731
G1 X102.0 Y119.0
G1 X104.731 Y118.731
G1 X107.358 Y117.934
G1 X109.778 Y116.641
G1 X111.899 Y114.899
G1 X113.641 Y112.778
G1 X114.934 Y110.358
G1 X115.731 Y107.731
G1 X116.0 Y105.0
G1 X115.731 Y102.269
G1 X114.934 Y99.6424
G1 X113.641 Y97.222
G1 X111.899 Y95.1005
G1 X109.778 Y93.3594
M5 S1
G0 Z3.0
G0 X125.066 Y99.6424
G0 Z1.0
G1 F150.0 Z-0.4
M3 S1000
G1 F110.0 X124.269 Y102.269
G1 X124.0 Y105.0
G1 X124.269 Y107.731
G1 X125.066 Y110.358
G1 X126.359 Y112.778
G1 X128.1 Y114.899
G1 X130.222 Y116.641
G1 X132.642 Y117.934
G1 X135.269 Y118.731
G1 X138.0 Y119.0
G1 X140.731 Y118.731
G1 X143.358 Y117.934
G1 X145.778 Y116.641
G1 X147.899 Y114.899
G1 X149.641 Y112.778
G1 X150.934 Y110.358
G1 X151.731 Y107.731
G1 X152.0 Y105.0
G1 X151.731 Y102.269
G1 X150.934 Y99.6424
G1 X149.641 Y97.222
G1 X147.899 Y95.1005
G1 X145.778 Y93.3594
G1 X143.358 Y92.0657
G1 X140.731 Y91.269
G1 X138.0 Y91.0
G1 X135.269 Y91.269
G1 X132.642 Y92.0657
G1 X130.222 Y93.3594
G1 X128.1 Y95.1005
G1 X126.359 Y97.222
G1 X125.066 Y99.6424
M5 S1
G0 Z3.0
G0 X160.187 Y103.1018
G0 Z1.0
G1 F150.0 Z-0.4
M3 S1000
G1 F110.0 X160.0 Y105.0
G1 X160.269 Y107.731
G1 X161.066 Y110.358
G1 X162.359 Y112.778
G1 X164.1 Y114.899
G1 X166.222 Y116.641
G1 X168.642 Y117.934
G1 X171.269 Y118.731
G1 X174.0 Y119.0
G1 X176.731 Y118.731
G1 X179.358 Y117.934
G1 X181.778 Y116.641
G1 X183.899 Y114.899
G1 X185.641 Y112.778
G1 X186.934 Y110.358
G1 X187.731 Y107.731
G1 X188.0 Y105.0
G1 X187.731 Y102.269
G1 X186.934 Y99.6424
G1 X185.641 Y97.222
G1 X183.899 Y95.1005
G1 X181.778 Y93.3594
G1 X179.358 Y92.0657
G1 X176.731 Y91.269
G1 X174.0 Y91.0
G1 X171.269 Y91.269
G1 X168.642 Y92.0657
G1 X166.222 Y93.3594
G1 X164.1 Y95.1005
G1 X162.359 Y97.222
G1 X161.066 Y99.6424
G1 X160.269 Y102.269
G1 X160.187 Y103.1018
M5 S1
G0 X0 Y0 Z0
M5 S1
G0 Z3.0
M30
`

gquinet, 2019-07-20:
And what i really don't understand is that i have another gcode file doing far more complicated and long job that works perfectly!

gquinet, 2019-07-20:
removing pololus did the same.
it stopped at position X:3.475   Y:2.956  Z:3.000

chamnit, 2019-07-21:
There is no such thing as an Alarm: 23. The alarm values only go from 1 to 9. 

Provide your Grbl $$ settings, $I build info, and any changes you made from the default build.

Also, make sure you are running the test job without anything connected to the Arduino except the USB cable. Not just the Pololu drivers.

gquinet, 2019-07-21:
settings:
`$$ < $0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=1600.000
$101=3200.000
$102=1600.000
$110=1000.000
$111=1000.000
$112=800.000
$120=30.000
$121=30.000
$122=30.000
$130=290.000
$131=360.000
$132=29.000
ok`

infos:
`$I < [VER:1.1f.20170801:]
[OPT:V,15,128]
ok`

test job given with the cnc works perfectly as some other jobs i made, .

tests i've made in my previous answer were as you said, without anything connected else to the board than the usb cable! (i tryed all kind of combination and the gcode stop always at the same point, just after starting on the first long move).

I use cambam to create gcode, is there some processor changes to make? maybe some instruction is not understood by the board? Does someone tried the gcode i posted on a camtool cnc v3.3 and have it working perfectly and entirely?

gquinet, 2019-07-21:
I just tried to replace G0 moves by G1 (except for Z moves) and it ran to the end without any problem!

this enforce my suspitions on the fast moves (G0)

chamnit, 2019-07-21:
You are exceeding the max supported step rate of 30 kHz. Reduce your micro stepping and or speed.

gquinet, 2019-07-21:
I was just thinking about the max speed too, but in grbl control, but you're right, i cannot reduce microstepping (wamtool cnc v3.3 does not have jumpers :D), so i reduced $111 to 500 (half the original value as i've doubled $131 as said before, and it looks to work now!

So it's not an issue on the board, i'll close it.

Thanks a lot.

closed 2019-07-21 by gquinet