gnea/grbl #691: disabled #define VARIABLE_SPINDLE does not produce expected results

ljubomirb, 2019-07-22:
I have some cheap Chinese CNC Laser board (2axes, but the one with two buttons, one power, other connected to some big resistor, for those who google to here).
The Arduino clone with it came empty, and no manual whatsoever was included.

When I burn GRBL 1.1 in, looking on oscilloscope, pins on cnc board connector do not react to g-code. Hooking oscilloscope to Arduino nano directly does show activity (as it should, so GRBL working perfect). After a lot of trial and error, I figured that I must have D11 and D12 switched board (or, pre GRBL 1.1 board). So I went and I comment out ` #define VARIABLE_SPINDLE`. However, after that, no activity on Arduino, directly on both D11 or D12 can be seen on scope.

Is there some other thing I should comment out, or is it possible that somehow GRBL no longer supports D12? It could, be that I have defective CPU, but, lets ignore this for now.

My plan is to use this for TTL laser, and I've already moved on to another board I have from before, but I recon, a lot of people might have this board, so if someone knows solutions, it will be a lot of help.

Thank you.

P.S.
not trying to lead you to wrong place, but since laser G-code is always generated with S command, could it be that pins do get swapped, but S is not parsed, as per:
```
#ifdef VARIABLE_SPINDLE
    printPgmString(PSTR(" S"));
    printFloat_RateValue(gc_state.spindle_speed);
  #endif
```

chamnit, 2019-07-22:
See the wiki explaining the pinout with variable spindle and without. Use the config.h file to alter features.You need variable spindle to use Grbl’s laser mode.

ljubomirb, 2019-07-22:
Hi chamnit, and thanks for fast reply!
I went by this statement from wiki:

> For Grbl v0.8 and v0.9+ with variable spindle disabled, Z-limit moves to D11 and spindle enable to D12. This is generally for backward compatibility on older boards.

The result on oscilloscope while the same g-code running in both cases is as follows:
-when enabled, D11 shows 0-5V series of on/offs, D12 is 5V
-when disabled, D11 shows 5V constant, and D12 is 5V from start to GRBL reset or M5 (no change during code run)

Can you, please, confirm that this is the normal result after disabling var spindle?

ljubomirb, 2019-07-25:
After not being able to make D12 work easily, I've given up playing try and error game with config.h and other parts. I realized there are simply too many ways to break things totally, so it's not worth it, especially cause you can buy board that does work out of the box cheap.
Thinking of it more, variable spindle config is **the only way to go**. Option to disable it only gives excuse to lazy board designers not to update their board designs for years.

So, this issue can be closed, as far as I'm concerned. Overall, I'm pretty happy how GRBL works, and my recommendations (that do not hold any weight at all) would then be rather:

- to remove this config option totally when some housekeeping is due
- to try focusing energy more towards leaving 328p (to make porting to other MCUs like STM32 more easy)
- to consider MIT licence

Great job!

chamnit, 2019-07-25:
D12 without variable spindle works fine. Removing this option will break compatibility for thousands of users with don’t use or need variable spindle.

If you hadn’t noticed, not much has changed to the Grbl code for the past couple of years. The reason for that is Grbl on the AVR is done. All efforts are dedicated to the next-gen version on ARM. This is not yet released to the public.

Grbl’s original GPL license makes it virtually impossible to switch to the MIT license. But in most cases, the GPL license hasn’t been a show stopper for most companies since GRbl is self contained on an microprocessor.

closed 2019-07-25 by chamnit

Shuhwaf, 2020-09-30:
i have designed a cnc router i a having one issue which i want you guys to help me with, when run the machine with the spindle disabled/ unconnected it will run without producing an alarm 1. if i connect and run the spindle the alarm 1(hard limit triggered) appears and the machine stops after running for a couple of minutes