gnea/grbl #696: [Feature] Digipots support

johnboiles, 2019-07-27:
I've got a [change in my branch](https://github.com/johnboiles/grbl-Mega-5X/commit/598eff46d0c8957fc9b53c9f7470bcfe28387ef2) to support Digipots for setting current on stepper drivers digitally. I added this to support my Ultimachine RAMBo 1.4 board. Is this something we'd want to include in the main `grbl` repo? I'm happy to make a PR to merge it in if so. Please advise!

chamnit, 2019-08-11:
Thanks! I think I'd be open to adding this to the Grbl Mega branch, as there is plenty of room to add this there.

johnboiles, 2019-08-11:
@chamnit great! So is the idea to keep this repo as minimal as possible and then add other features to processor-specific forks? Should I do the same for #695? I’ll of course keep everything behind `#define`s so that it can be compiled out for people who don’t need it

johnboiles, 2019-08-15:
@chamnit I put up my pull-request here: https://github.com/gnea/grbl-Mega/pull/108

One thing for us to consider, is that if we're adding a per-axis setting to gnea/grbl-Mega and not gnea/grbl, we'll probably want to at least reserve those settings codes in gnea/grbl so that we don't diverge on which setting codes do what.

Alternately, we could probably put this into gnea/grbl, since it's fully compiled out if `HAS_DIGIPOTS` is not defined.

What do you think?