gnea/grbl #697: Clobbering pull-up resistors.

bdurbrow, 2019-07-28:
The method used to set bits on the IO ports clobbers any pull-up resistor settings for other bits on the port.
Setting multiple bits simultaneously with a single statement results in machine code being generated that modifies unintended bits.

Suppose bit 7 on port A is being used as an input, with a pull-up; and bits 2 and 4 are being used as an output, initialized like so:
```
DDRA = (1<<2) | (1<<4);
PORTA = (1<<7);
```

OK, now some external hardware pulls pin A7 low.

Then, while it's being held low, bit values for A2 & A4 is set (high or low doesn't matter... but I'll show the high case).
`PORTA |= ((1<<2) | (1<<4));`

This results in machine code that reads the port; does a bitwise or with the value of ((1<<2) | (1<<4)); and writes that back. This will occur even if it's a variable being used to store which bits to turn on, and the variable only ever has one bit turned on in it.

The problem is that the value for bit 7 was read as low, and writing that back results in disabling the pull-up resistor on that pin.

To avoid this the bits for the pins need to be written individually:
```
PORTA |= (1<<2);
PORTA |= (1<<4);
```

which compiles to:

```
  sbi 34-0x20,2
  sbi 34-0x20,4

```
For variables an if statement  must be used:
```
  if(q & (1<<2)) PORTA |= 1<<2; else PORTA &=~ (1<<2);
  if(q & (1<<4)) PORTA |= 1<<4; else PORTA &=~ (1<<4);

```

Beware that using an ?: operator and then using the result value of the operator generates machine code that has this problem:

```
uint8_t foo(uint8_t q)
{
    return (q & (1<<3))?(PORTA |= 1<<3):(PORTA &=~ (1<<3));
}
```

compiles to:
```
foo:
  sbrs r24,3
  rjmp .L2
  in r24,34-0x20
  ori r24,lo8(8)
  out 34-0x20,r24
  ret
.L2:
  in r24,34-0x20
  andi r24,lo8(-9)
  out 34-0x20,r24
  ret

```
Note the separate in, ori, and out instructions generated.

Also, a related issue was spotted:

STEP_PULSE_DELAY potentially clobbers any bits that are also on the port(s) that change state between the main ISR(TIMER1_COMPA_vect) firing and the delayed ISR(TIMER0_COMPA_vect) firing.

1) The main ISR stores the value of the entire port into st.step_bits:
`      st.step_bits = (STEP_PORT & ~STEP_MASK) | st.step_outbits; // Store out_bits to prevent overwriting.`
2) Some other event happens (ISR firing after the sei() call?) and one of the port bits not allocated to step outputs changes state.
3) The delayed ISR(TIMER0_COMPA_vect) fires, and reads the old value of the bit(s) in question from st.step_bits.
4) The bits are then clobbered by being written to by the ISR.
`      STEP_PORT = st.step_bits; // Begin step pulse.`

... I think the proper way to do this is to not store the entire port value, but just the step bits that should be pulsed. Also, care must be excersized such that the code that is used resolves to sbi & cbi instructions so that any pull-ups on the port are not accedentally enabled or disabled.

1) In main ISR(TIMER1_COMPA_vect):
`      st.step_bits = st.step_outbits; // Store out_bits to for delay.`
2) In the delayed ISR(TIMER0_COMPA_vect):
```
      if(st.step_bits & (1<<STEP_BIT(X_AXIS))) STEP_PORT |= (1<<STEP_BIT(X_AXIS)); else STEP_PORT &= ~(1<<STEP_BIT(X_AXIS));
      if(st.step_bits & (1<<STEP_BIT(Y_AXIS))) STEP_PORT |= (1<<STEP_BIT(Y_AXIS)); else STEP_PORT &= ~(1<<STEP_BIT(Y_AXIS));
      if(st.step_bits & (1<<STEP_BIT(Z_AXIS))) STEP_PORT |= (1<<STEP_BIT(Z_AXIS)); else STEP_PORT &= ~(1<<STEP_BIT(Z_AXIS));

```
For the non-delayed case, obviously, just set the bits like so:

```
      if(st.step_outbits & (1<<STEP_BIT(X_AXIS))) STEP_PORT |= (1<<STEP_BIT(X_AXIS)); else STEP_PORT &= ~(1<<STEP_BIT(X_AXIS));
      if(st.step_outbits & (1<<STEP_BIT(Y_AXIS))) STEP_PORT |= (1<<STEP_BIT(Y_AXIS)); else STEP_PORT &= ~(1<<STEP_BIT(Y_AXIS));
      if(st.step_outbits & (1<<STEP_BIT(Z_AXIS))) STEP_PORT |= (1<<STEP_BIT(Z_AXIS)); else STEP_PORT &= ~(1<<STEP_BIT(Z_AXIS));
```

According to the Compiler Explorer at godbolt.org, this produces something like this:

```
sbrs r24,3
  rjmp .L2
  sbi 34-0x20,3
  rjmp .L3
.L2:
  cbi 34-0x20,3
```
  
... which shouldn't clobber any bits.

FWIW, I am working to patch my fork of grbl-mega to address this.

cri-s, 2019-07-28:
That don't make sense as PORTA is the output latch of the port. The
behavior you describe exist with pic's as example as this have a single
port register where output latch and input read register shares the same
address. On avr or Arm or .... Where this registers have different
addresses this write-read-modify don't exists.

bdurbrow, 2019-07-28:
On AVR, they **_DO_** share the same address. PORTA is PORTA, reading or writing.

I just stuck this:

`PORTA |= 0x03;`

in Compiler Explorer, and got this:

```
in r25,34-0x20
ori r25,lo8(3)
out 34-0x20,r25
```
Note that the address being read from and written to (34-0x20) is the same.

Also, 
`uint8_t a=PORTA;`
generates
`in r25,34-0x20`
... same address.

From the datasheet for the ATMega2560 (it's the same setup as on the ATMega328p, but I'm working on the arduino mega port of grbl... so it's what I have handy)

> Each port pin consists of three register bits: DDxn, PORTxn, and PINxn.

> the DDxn bits are accessed at the DDRx I/O address, the PORTxn bits at the PORTx I/O address, and the PINxn bits at the PINx I/O address.

> The DDxn bit in the DDRx Register selects the direction of this pin. If DDxn is written logic one, Pxn is configured as an output pin. If DDxn is written logic zero, Pxn is configured as an input pin.
> If PORTxn is written logic one when the pin is configured as an input pin, the pull-up resistor is activated. To switch the pull-up resistor off, PORTxn has to be written logic zero or the pin has to be configured as an output pin. The port pins are tri-stated when reset condition becomes active, even if no clocks are running.
> If PORTxn is written logic one when the pin is configured as an output pin, the port pin is driven high (one). If PORTxn is written logic zero when the pin is configured as an output pin, the port pin is driven low (zero).

cri-s, 2019-07-28:
On Avr, reading a port (PINx) have a different address as writing a port
(PORTx). There exist MCU that have just a single address for reading and
writing a gpio port. Having different addresses the port is just a latch
that never can be influenced by external pin state.
Because of this the problematic you try to adress is not present on avr/arm
MCU.

bdurbrow, 2019-07-28:
Doh! Well, that’s what I get for posting before breakfast. :flushed:
Re-reading the data sheet with some food & caffeine in me indicates that your right about the first part.

I still _think_ the second part is valid... reading the PORTx register and then writing it back from a different ISR should introduce a race condition...

cri-s, 2019-07-28:
The foo example you made is wrong.
Here you have a big coding problem as port variables MUST and ARE declared
as volatile. foo is not qualified to operate on port vars, and further
volatile as function argument have different behavior as volatile inside
code, as volatile as function argument could be cached or aliases where
that is forbidden to  plain volatile code

The delayed pulse is made for speed reason. As that register could be
preallocated to dedicated register, the isr could be set to naked in order
it need only 5 CPU cycles.
As the output port is near exclusive for output, it don't matter. The only
problem is with arduino uno/micro/... and stepper enable pin. If that pin
was a input, the problem was inexistent, but sw workaround exist.
st_shutdown() must wait until TCCR0B is 0 or it must set it to 0 after a
timeout.
This is only needed for slow machines where step delay is used with avr328 .

bdurbrow, 2019-07-28:
> The foo example you made is wrong.

I get that my earlier interpretation was erroneous. Now, pardon me if I'm being dense, but... I don't quite get how foo is somehow unqualified to operate on port variables?

Given:
```
uint8_t foo(uint8_t q)
{
    return (q & (1<<3))?(PORTA |= 1<<3):(PORTA &=~ (1<<3));
}
```
q doesn't need to be volatile; as no other thread/ISR can access it.

FWIW, feeding this:

```
#define __AVR_ATmega2560__
#include <avr/io.h>
#include <stdint.h>
uint8_t foo(volatile uint8_t q)
{
    return (q & (1<<3))?(PORTA |= 1<<3):(PORTA &=~ (1<<3));
}

uint8_t bar(uint8_t q)
{
    return (q & (1<<3))?(PORTA |= 1<<3):(PORTA &=~ (1<<3));
}
```

to Compiler Explorer generates this:

```
__SP_H__ = 0x3e
__SP_L__ = 0x3d
__tmp_reg__ = 0
foo:
  push r28
  push r29
  push __tmp_reg__
  in r28,__SP_L__
  in r29,__SP_H__
  std Y+1,r24
  ldd r24,Y+1
  sbrs r24,3
  rjmp .L2
  in r24,34-0x20
  ori r24,lo8(8)
  rjmp .L4
.L2:
  in r24,34-0x20
  andi r24,lo8(-9)
.L4:
  out 34-0x20,r24
  pop __tmp_reg__
  pop r29
  pop r28
  ret
bar:
  sbrs r24,3
  rjmp .L6
  in r24,34-0x20
  ori r24,lo8(8)
  out 34-0x20,r24
  ret
.L6:
  in r24,34-0x20
  andi r24,lo8(-9)
  out 34-0x20,r24
  ret
```

I think the reason it's doing what it is in the volatile function argument case (moving q from r24 to the stack), is so that q can - in principal, at least - be accessed elsewhere... even though it happens to be that there's no code here that makes that stack location available elsewhere.

___________________________________________________________________________________________

> The delayed pulse is made for speed reason. As that register could be
preallocated to dedicated register, the isr could be set to naked in order
it need only 5 CPU cycles.

That... does not appear to actually be happening?

Disassembly of the actual compiled .elf with the step pulse delay turned on:
```

0000501a <__vector_21>:
  // initiated after the STEP_PULSE_DELAY time period has elapsed. The ISR TIMER2_OVF interrupt
  // will then trigger after the appropriate settings.pulse_microseconds, as in normal operation.
  // The new timing between direction, step pulse, and step complete events are setup in the
  // st_wake_up() routine.
  ISR(TIMER0_COMPA_vect)
  {
    501a:	1f 92       	push	r1
    501c:	0f 92       	push	r0
    501e:	0f b6       	in	r0, 0x3f	; 63
    5020:	0f 92       	push	r0
    5022:	11 24       	eor	r1, r1
    5024:	0b b6       	in	r0, 0x3b	; 59
    5026:	0f 92       	push	r0
    5028:	8f 93       	push	r24
    502a:	ef 93       	push	r30
    502c:	ff 93       	push	r31
    #ifdef DEFAULTS_RAMPS_BOARD
      STEP_PORT(0) = st.step_bits[0]; // Begin step pulse.
    502e:	ee ef       	ldi	r30, 0xFE	; 254
    5030:	fc e0       	ldi	r31, 0x0C	; 12
    5032:	84 85       	ldd	r24, Z+12	; 0x0c
    5034:	81 bb       	out	0x11, r24	; 17
      STEP_PORT(1) = st.step_bits[1]; // Begin step pulse.
    5036:	85 85       	ldd	r24, Z+13	; 0x0d
    5038:	81 bb       	out	0x11, r24	; 17
      STEP_PORT(2) = st.step_bits[2]; // Begin step pulse.
    503a:	86 85       	ldd	r24, Z+14	; 0x0e
    503c:	80 93 0b 01 	sts	0x010B, r24	; 0x80010b <__TEXT_REGION_LENGTH__+0x70010b>
    #endif // Ramps Board
  }
    5040:	ff 91       	pop	r31
    5042:	ef 91       	pop	r30
    5044:	8f 91       	pop	r24
    5046:	0f 90       	pop	r0
    5048:	0b be       	out	0x3b, r0	; 59
    504a:	0f 90       	pop	r0
    504c:	0f be       	out	0x3f, r0	; 63
    504e:	0f 90       	pop	r0
    5050:	1f 90       	pop	r1
    5052:	18 95       	reti
```

Again... pardon me if I'm being dense, but... I don't see how doing the load & mask operation in the delayed ISR has a significant impact on performance: the main ISR gets shorter by ~ 5 instructions, and the delayed ISR gets longer by ~ 5 instructions (or so I gather from disassembling the compiler's output). Yes, that does affect the accuracy of the delay... but I don't think it has a harmful effect - generally, if a delay is needed, a slightly longer one should be OK?

> As the output port is near exclusive for output, it don't matter. 

That's assuming standard pinout configuration. FWIW, I think, moving forwards (grbl-mega; grbl on ARM) that having it be technically correct at the small expense of a little latency on the step delay is preferable. I certainly can understand the '328 version never getting a change here; as there's no room left on the chip for something that _could_ change pin state between ISRs.

:smile:

cri-s, 2019-07-28:
inline

> I get that my earlier interpretation was erroneous. Now, pardon me if I'm
> being dense, but... I don't quite get how foo is somehow unqualified to
> operate on port variables?
>
> Given:
> ```
> uint8_t [...]

Sorry, it was my fault, i was thinking that foo uses pointer and not
jusr args, sorry.
> ```
> #define __AVR_ATmega2560__
> #include <avr/io.h>
> #include <stdint.h>
> uint8_t foo(volatile uint8_t q)
> {
>     return (q & (1<<3))?(PORTA |= 1<<3):(PORTA &=~ (1<<3));
> }
>
> uint8_t bar(uint8_t q)
> {
>     return (q & (1<<3))?(PORTA |= 1<<3):(PORTA &=~ (1<<3));
> }
> ```
>
> to Compiler Explorer generates this:
>
> ```
> __SP_H__ = 0x3e
> __SP_L__ = 0x3d
> __tmp_reg__ = 0
> foo:
>   push r28
>   push r29
>   push __tmp_reg__
>   in r28,__SP_L__
>   in r29,__SP_H__
>   std Y+1,r24
>   ldd r24,Y+1
>   sbrs r24,3
>   rjmp .L2
>   in r24,34-0x20
>   ori r24,lo8(8)
>   rjmp .L4
> .L2:
>   in r24,34-0x20
>   andi r24,lo8(-9)
> .L4:
>   out 34-0x20,r24
>   pop __tmp_reg__
>   pop r29
>   pop r28
>   ret
> bar:
>   sbrs r24,3
>   rjmp .L6
>   in r24,34-0x20
>   ori r24,lo8(8)
>   out 34-0x20,r24
>   ret
> .L6:
>   in r24,34-
>   andi r24,lo8(-9)
>   out 34-0x20,r24
>   ret
> ```

Just to compare generated code,
[... 125 lines cut ...]
    39a2:       0e 94 18 50     call    0xa030  ; 0xa030 <hal_WriteSteps>
    39a6:       15 bc           out     0x25, r1        ; 37
    39a8:       ff 91           pop     r31
    39aa:       ef 91           pop     r30
    39ac:       bf 91           pop     r27
    39ae:       af 91           pop     r26
    39b0:       9f 91           pop     r25
    39b2:       8f 91           pop     r24
    39b4:       7f 91           pop     r23
    39b6:       6f 91           pop     r22
    39b8:       5f 91           pop     r21
    39ba:       4f 91           pop     r20
    39bc:       3f 91           pop     r19
    39be:       2f 91           pop     r18
    39c0:       0f 90           pop     r0
    39c2:       0b be           out     0x3b, r0        ; 59
    39c4:       0f 90           pop     r0
    39c6:       0f be           out     0x3f, r0        ; 63
    39c8:       0f 90           pop     r0
    39ca:       1f 90           pop     r1
    39cc:       18 95           reti

Nothing wrong with that, but, it consumes unnecessary 3us of time.
As two interrupts are fired inside the main interrupt code, the are
6us of wasted time.

closed 2019-08-11 by chamnit