gnea/grbl #700: End Stop Oddity

3boysdad, 2019-08-03:
Have a xPro V3 runningGRBL v1.1f.

When I have these settings:

$0=10
$1=255
$2=0
$3=3
$4=1
$5=0
$6=0
$10=19
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=3

I can trigger and see that the end stops are working by executing the status now command ?

<Idle|MPos:0.000,0.000,0.000|Bf:15,127|FS:0,0|Pn:Y|WCO:-6.440,-56.640,0.000>
<Idle|MPos:0.000,0.000,0.000|Bf:15,127|FS:0,0|Pn:X|WCO:-6.440,-56.640,0.000>

But when I turn on hard stops:

$0=10
$1=255
$2=0
$3=3
$4=1
$5=0
$6=0
$10=19
$11=0.020
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3

My board goes triggers an alarm...

ALARM:1
[MSG:Reset to continue]

I have some standard end stops where I have the NO pin wired up to the switch port and ground/COM wired up to the ground pin on the board.  I can tell - when the hard end stops is "off" that they are triggering.  But when it's on - why does it fire an alarm?  Also the homing flat out doesn't do anything all - just sits there.

chamnit, 2019-08-03:
You’re having some EMI issues with your limit switches. It’s a very common problem due to inadequate shielding, grounding, etc. Try searching for solutions. There are lots

closed 2019-08-03 by chamnit

3boysdad, 2019-08-03:
i see...yet they trigger correctly, and not cause the board to alarm, while the setting is 'off'.  wound't the interference occur _regardless_ of the $21=1/0 setting?  current, resistance, and signaling, doesn't change when $21 is toggled.