gnea/grbl #701: spindle spin up time.

J-Dunn, 2019-08-05:
I was just reading about grbl laser mode and it explains that std grbl mode waits each time spindle speed it changed to allow time for spin up.  I'm not seeing this. 

I have variable spindle speed on PWM and spindle dir as enable option. 

When I start a job which includes S24000M3 , the spindle starts its acceleration and grbl just seems to move off straight away and run the rest of the GCODE.   Am I mistaken about when this happens ( do I need to start slow and then alter speed when already in M3 state ? ). 

Is there  a way to insert a pause or length the default to be more noticeable?  For 24000 rpm I need 10 or 15s  to get up to full speed.

barma1ey, 2019-08-06:
Try use G4Px command  (where x is time in seconds) before setting starting spindle.

J-Dunn, 2019-08-06:
Thanks but I don't see G4 in the list of GRBL commands, is it supported?
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

barma1ey, 2019-08-06:
I've got no idea why G4 is not listed but it is totally supported

chamnit, 2019-08-06:
It’s listed in the table half a page down.

J-Dunn, 2019-08-07:
Many thanks, strange I overlooked that, I even did a search for G4 and only found the G40s. 

If I send G20 G4 S5 then S24000M3X-6 it moves off directly.  I was expecting a 5s wait, where the spindle could speed up, then the move. 

Am I misunderstanding what G4 is about ?

barma1ey, 2019-08-07:
You should set up spindle speed and direction prior to delay command.
for instance:
S15000M3
G4P10 (wait for 10 sec)
X+100500

J-Dunn, 2019-08-07:
Duh, I don't know why I thought the param was S instead of P , 
Putting in valid code makes it work a lot better. 

Must be getting too much sun this year. 

Many thanks for unclogging my brain.

chamnit, 2019-08-11:
@J-Dunn : Could you point me to where in the Wiki you see that Grbl waits to spin up the spindle prior to moving? I'd like to fix that. By default, Grbl forces a stop in motion when spindle speed or state is changed, but it doesn't wait, except in a few circumstances. If I recall correctly, it waits after a parking motion so it doesn't plunge or resume cutting before the spindle is up to speed. Otherwise, you do have to program a G4 during a job.

J-Dunn, 2019-08-11:
I don't think I was saying I had seen that. So far I have only tested setting spindle speed before motion starts, so it will be at stop when spindle state is changed. It then moves off directly if I do not add a dwell command. My whole problem here was correctly sending a G4 at the right point to get a pause for spin-up time.  I'm not adding a G4 on the same line as M3 and I get required result.

`M3S24000G4P15`

All seems to be working correctly on the GRBL end so I'll close this issue. Thanks.

closed 2019-08-11 by J-Dunn